autopower/thermeq3 #5: thermeq3 - control of Max! wireless thermostatic radiator valves

dominator99, 2015-12-06:
Anyone had any experience of this software as I'm not getting any response to PM's from author

https://github.com/autopower/thermeq3

autopower, 2015-12-06:
Hello, feel free to post message on FB profile https://www.facebook.com/autopow/ as I don't see any message from you here.

closed 2015-12-06 by autopower

dominator99, 2015-12-06:
Here's my original response to your Arduino forum post:
Hi autopower

just came across this posting & it looks really interesting.

I've just aquired an Arduino Yun, Max! cube & 5 wireless radiator thermostats & I'm interested in controlling my gas boiler as per your post.

I currently control a gas boiler/heat pump system using an Arduino Mega 2560 with 13 DS18B20 digital temps sensors. Two of these sensors replaces my 'normal' roomstat (one measures room temp from inside roomstat housing & the other at floor level. A third sensor measures outside air temp. I use an algorithm to adjust the required room temp constantly in real time & 'fire' the gas boiler or heat pump accordingly.

Unfortunately, these 2 room temp sensors are located in a South facing living room which experiences solar gain meaning the CH can be prevented from coming on even when other North facing rooms need heating.

Your system seems to offer a way around this problem by using individual or 'sum of' thermostat valve opening to turn the CH on, overriding the 2 temp sensors.

My big problem is understanding your system (I'm a retired mechanical engineer & my coding is limited to simple Arduino IDE)

" python-openssl library `opkg update; opkg install python-openssl`" - I don't understand this; can you explain (simple please)

"python code, upload into /root/ files nsm.py and config.py " - I run my Arduino sketches under Windows 7 so how can I copy the 2 files above into the /root folder ?
I have Ubuntu installed on my Win 7 as a virtual machine if that's of any use.

autopower, 2015-12-06:
Hello,
* opkg issue: run this command on yun, so you must telnet/ssh to your yun as root
* python code in /root, copy files to /root directory of your yun
nsm.py is closely coupled with arduino yun, you don't need any linux distro on your local machine (Win7). 
I'll write down more comprehensive readme soon.

dominator99, 2015-12-06:
Thanks for that.  Look forward to your new readme

dominator99, 2015-12-11:
Hi Michal

I copied files nsm.py and config.py into the root of my Yun SD card; is 
this correct?

I uploaded your sketch to the Yun.

I changed the ip address in config.py to my Max! cube to 192.168.0.222 
(as detailed on the back of the cube) using:

192.168.0.29/data/put/stp.max_ip = 192.168.0.222

192.168.0.29 is the address of my Yun

When I boot the Yun I do not see the flashing LED you mention.

When I run the serial monitor I get error messages:

Unable to connect: retrying (1)...
Unable to connect: retrying (2)...
Unable to connect: retrying (3)...
Unable to connect: retrying (4)...
Unable to connect: is the sketch using the bridge?

Please help!

Regards

Chris

autopower, 2015-12-12:
Hello,
to change ip address in your config.py just edit file to look like:
```
stp.max_ip = "192.168.0.222"
stp.fromaddr = "devices@foo.local"
stp.toaddr = "user@foo.local"
stp.mailserver = "mail.foo.local"
stp.mailport = 25
stp.frompwd = "this.is.password"
stp.devname = "hellmostat"
stp.timeout = 10
```

dominator99, 2015-12-12:
Hi again Michal

I managed to figure that out yesterday & now get the 4 sec red & green 
LED flashing referred to in your readme.
I also get the green LED flashing when the 'bridge' is established; - so 
far so good!

I've wired up a breadboard as per your instructions except I'm 
simulating a 5V relay with a yellow LED.

I changed the value of :-   "valve": ["valve_pos", 35],   to "valve": 
["valve_pos", 5],
                                         &  "valves": ["valves", 1],   
to      "valves": ["valves", 5], in file nsm.py (I have 5 valves)
by manually editing the nsm.py file using Wordpad & restarting your 
Arduino sketch 'thermeq3.ino' but can't get my '5V relay' yellow LED to 
light with at least 1 valve open?

I also noticed that some of the names used in README.md don't appear in 
nsm.py

for example:

Heating also can be started if sum of radiator valves positions are 
geater than `num_of_valves * stp.per_switch`,
where `stp.per_switch` is value in %. So if you have 10 valves in house 
and `stp.per_switch=8`, and sum of these
valves positions are 80+, relay is switched on. You can turn on this 
feature by `stp.preference="per"` in python code.
If you need only simple total value, set `stp.preference="total"` and 
setup `total_switch` variable.

I assume "valves": ["valves", 1]  is used for 1 valve not num_of_valves

I can't find stp.per_switch`,  or  `stp.per_switch` is this "valve": 
["valve_pos", 35] meaning valve open 35%?

I haven't checked for any other typos as I'm still bogged down at the 
start but everything seems to be going in the right direction; albeit 
slowly.

PS remember I'm a retired professional mechanical engineer not a coder!! :)

Thanks for your interest in helping - I'm surprised nobody else has 
joined the discussion of these wireless thermostats!

Regards

Chris

dominator99, 2015-12-12:
Hi again Michal

I managed to figure that out yesterday & now get the 4 sec red & green 
LED flashing referred to in your readme.
I also get the green LED flashing when the 'bridge' is established; - so 
far so good!

I've wired up a breadboard as per your instructions except I'm 
simulating a 5V relay with a yellow LED.

I changed the value of :-   "valve": ["valve_pos", 35],   to "valve": 
["valve_pos", 5],
                                         &  "valves": ["valves", 1],   
to      "valves": ["valves", 5], in file nsm.py (I have 5 valves)
by manually editing the nsm.py file using Wordpad & restarting your 
Arduino sketch 'thermeq3.ino' but can't get my '5V relay' yellow LED to 
light with at least 1 valve open?

I also noticed that some of the names used in README.md don't appear in 
nsm.py

for example:

Heating also can be started if sum of radiator valves positions are 
geater than `num_of_valves * stp.per_switch`,
where `stp.per_switch` is value in %. So if you have 10 valves in house 
and `stp.per_switch=8`, and sum of these
valves positions are 80+, relay is switched on. You can turn on this 
feature by `stp.preference="per"` in python code.
If you need only simple total value, set `stp.preference="total"` and 
setup `total_switch` variable.

I assume "valves": ["valves", 1]  is used for 1 valve not num_of_valves

I can't find stp.per_switch`,  or  `stp.per_switch` is this "valve": 
["valve_pos", 35] meaning valve open 35%?

I haven't checked for any other typos as I'm still bogged down at the 
start but everything seems to be going in the right direction; albeit 
slowly.

PS remember I'm a retired professional mechanical engineer not a coder!! :)

Thanks for your interest in helping - I'm surprised nobody else has 
joined the discussion of these wireless thermostats!

PPS  In your README.md file you quote:

##How to debug?
Python code produce 3 files:
* `/mnt/sd<x1>/<device_name>.csv`, simple comma separated value file 
with valve positions and temperature readings
* `/mnt/sd<x1>/<device_name>.log`, log file, really huge on `log_debug`
* `/mnt/sd<x1>/<device_name>_error.log`, python stderr redirected, use 
in case of crash, or send me this file.
* `/mnt/sd<x1>/<device_name>.bridge` saved bridge client values
* `/root/nsm.error` low level errors, which cant be written do .log file 
(e.g. due to lack of mounted storage media)
This file is also mailed to recipient on start. Then is truncated to 
zero size.

Where are these 3 files stored? I couldn't find them on my SD card or 
are they hidden?

Regards

Chris

autopower, 2015-12-12:
Hello Chris,
yes, readme is little bit outdated. I'll post update, meanwhile:
* valve_pos is valve position in % when heating is started, in "per" mode
* valves is number how many valves must be over valve_pos to start heating, in "per" mode
* total_switch is similar to valve_pos but for sum of all valves, e.g. if sum is over total_switch then heating is switched on, you must be in "total" mode
* mode is switched with "preference"="per" or ="total"
* log files are in /mnt/sda1 or /mnt/sdb1, to clarify type this command `cd /mnt|ls -al` and post results here.

dominator99, 2015-12-13:
Hi Michal

Had a nightmare 2 days with the Yun where it bricked itself for no 
apparent reason; factory reset wouldn't work but after a lot of Googling 
managed to reset it.

When I run your thermeq3 sketch after about 10 secs I get the disco 
light scene.

I've attached 2 logs  & bridge to see if you can shed any light on the 
problem; all the csv files are empty

Regards

Chris

16:09:54 Redirection active
Traceback (most recent call last):
  File "/root/nsm.py", line 1526, in <module>
    doLoop()
  File "/root/nsm.py", line 1440, in doLoop
    time.sleep(stp.intervals["slp"][0])
KeyboardInterrupt
16:41:36 Redirection active
Traceback (most recent call last):
  File "/root/nsm.py", line 1526, in <module>
    doLoop()
  File "/root/nsm.py", line 1399, in doLoop
    if tryRead("beta", "no", False).upper() == "YES":
AttributeError: 'NoneType' object has no attribute 'upper'

2015/12/13 15:50:20 - thermeq3 - INFO - --> V139 started with PID=1851 <--
2015/12/13 15:50:29 - thermeq3 - ERROR - Error loading bridge file, using defaults!
2015/12/13 15:50:29 - thermeq3 - DEBUG - Queqing [S]
2015/12/13 15:50:29 - thermeq3 - DEBUG - Message queue=['S']
2015/12/13 15:50:29 - thermeq3 - DEBUG - Sending message [S]
2015/12/13 15:50:51 - thermeq3 - ERROR - Error opening connection to MAX Cube. Error: [Errno 148] No route to host
2015/12/13 15:50:51 - thermeq3 - ERROR - Traceback: Traceback (most recent call last):
  File "/root/nsm.py", line 796, in openMAX
    var.client_socket.connect((stp.max_ip, 62910))
  File "/usr/lib/python2.7/socket.py", line 224, in meth
    return getattr(self._sock,name)(*args)
error: [Errno 148] No route to host

2015/12/13 15:50:51 - thermeq3 - DEBUG - No key 000000 in d_W. Key added.
2015/12/13 15:50:53 - thermeq3 - ERROR - Error connecting to mail server mail.foo.local:25. Error code: [Errno -2] Name or service not known
2015/12/13 15:50:53 - thermeq3 - ERROR - Traceback: Traceback (most recent call last):
  File "/root/nsm.py", line 382, in sendEmail
    server = smtplib.SMTP(stp.mailserver, stp.mailport)
  File "/usr/lib/python2.7/smtplib.py", line 249, in __init__
    (code, msg) = self.connect(host, port)
  File "/usr/lib/python2.7/smtplib.py", line 309, in connect
    self.sock = self._get_socket(host, port, self.timeout)
  File "/usr/lib/python2.7/smtplib.py", line 284, in _get_socket
    return socket.create_connection((port, host), timeout)
  File "/usr/lib/python2.7/socket.py", line 553, in create_connection
    for res in getaddrinfo(host, port, 0, SOCK_STREAM):
gaierror: [Errno -2] Name or service not known

2015/12/13 15:50:53 - thermeq3 - DEBUG - Queqing [E]
2015/12/13 15:50:53 - thermeq3 - DEBUG - Message queue=['E']
2015/12/13 15:50:54 - thermeq3 - DEBUG - Sending message [E]
2015/12/13 15:50:54 - thermeq3 - DEBUG - Creating new daily key: 13-12-2015
2015/12/13 15:50:54 - thermeq3 - DEBUG - Day reading warnings value=0
2015/12/13 15:50:59 - thermeq3 - DEBUG - Bridge file saved.
2015/12/13 15:50:59 - thermeq3 - INFO - Zero sized stderr log file, nothing'll be send
2015/12/13 15:50:59 - thermeq3 - ERROR - Unable to get latest version info - URLError = unknown url type: https
2015/12/13 15:51:05 - thermeq3 - DEBUG - Bridge file saved.
2015/12/13 15:51:05 - thermeq3 - DEBUG - Public IP address: 86.6.59.107
2015/12/13 15:51:06 - thermeq3 - DEBUG - OWW interval updated to [720, 2160, 2160]
[... 743 lines cut ...]
2015/12/13 17:26:50 - thermeq3 - DEBUG - Queqing [E]
2015/12/13 17:26:50 - thermeq3 - DEBUG - Message queue=['E']
2015/12/13 17:26:50 - thermeq3 - DEBUG - Sending message [E]

openwinlist={}
ignore_opened=45
valve_pos=5
interval=90
autoupdate=True
error=69
total_switch=150
uptime=0:26:58
current_status={}
preference=per
msg=
status=defaults
heattime={"total": [0, 0.0], 				{datetime.datetime.date(datetime.datetime.now()).strftime("%d-%m-%Y"): [0, time.time()]}
beta=no
heattime_string=0:00:00
totalerrors=0
app_uptime=0:21:07
command=
daily=
no_oww=0
valves=1

autopower, 2015-12-13:
Please download new version of [nsm.py](https://github.com/autopower/thermeq3/blob/master/install/nsm.py) and replace oldone wih this version. To copy files to yun you can use for example [winscp](https://winscp.net/eng/index.php).

dominator99, 2015-12-20:
Hi Michal

Please find attached latest error & status reports for your perusal & 
advice where I'm going wrong.

I currently have a flashing green (loop) LED & a permanently lit red 
(error) LED

-- 

Regards

Chris

2015/12/20 15:07:40 - thermeq3 - INFO - --> V146 started with PID=13554 <--
2015/12/20 15:15:56 - thermeq3 - INFO - --> V146 started with PID=15413 <--
2015/12/20 15:25:56 - thermeq3 - INFO - --> V146 started with PID=17655 <--
2015/12/20 15:35:56 - thermeq3 - INFO - --> V146 started with PID=19905 <--
2015/12/20 15:45:56 - thermeq3 - INFO - --> V146 started with PID=22147 <--
2015/12/20 15:55:56 - thermeq3 - INFO - --> V146 started with PID=24396 <--
2015/12/20 16:05:56 - thermeq3 - INFO - --> V146 started with PID=26638 <--
2015/12/20 16:15:56 - thermeq3 - INFO - --> V146 started with PID=28887 <--
2015/12/20 16:25:56 - thermeq3 - INFO - --> V146 started with PID=31129 <--
2015/12/20 16:35:56 - thermeq3 - INFO - --> V146 started with PID=924 <--
2015/12/20 16:45:56 - thermeq3 - INFO - --> V147 started with PID=3312 <--
2015/12/20 16:46:03 - thermeq3 - ERROR - Bridge error codeword [heattime] value [{"total": [0, 0.0], 				{datetime.datetime.date(datetime.datetime.now()).strftime("%d-%m-%Y"): [0, time.time()]}]
2015/12/20 16:46:05 - thermeq3 - DEBUG - Bridge file loaded.
2015/12/20 16:46:05 - thermeq3 - DEBUG - Creating new daily key: 20-12-2015
2015/12/20 16:46:05 - thermeq3 - DEBUG - Day reading warnings value=0
2015/12/20 16:46:11 - thermeq3 - DEBUG - Bridge file saved.
2015/12/20 16:46:11 - thermeq3 - DEBUG - Queqing [S]
2015/12/20 16:46:11 - thermeq3 - DEBUG - Message queue=['S']
2015/12/20 16:46:11 - thermeq3 - DEBUG - Sending message [S]
2015/12/20 16:46:28 - thermeq3 - ERROR - Error opening connection to MAX Cube. Error: [Errno 146] Connection refused
2015/12/20 16:46:28 - thermeq3 - ERROR - Traceback: Traceback (most recent call last):
  File "/root/nsm.py", line 882, in openMAX
    var.client_socket.connect((stp.max_ip, 62910))
  File "/usr/lib/python2.7/socket.py", line 224, in meth
    return getattr(self._sock,name)(*args)
error: [Errno 146] Connection refused

2015/12/20 16:46:28 - thermeq3 - DEBUG - No key 000000 in d_W. Key added.
2015/12/20 16:46:28 - thermeq3 - ERROR - Error connecting to mail server mail.foo.local:25. Error code: [Errno -2] Name or service not known
2015/12/20 16:46:28 - thermeq3 - ERROR - Traceback: Traceback (most recent call last):
  File "/root/nsm.py", line 442, in sendEmail
    server = smtplib.SMTP(stp.mailserver, stp.mailport)
  File "/usr/lib/python2.7/smtplib.py", line 249, in __init__
    (code, msg) = self.connect(host, port)
  File "/usr/lib/python2.7/smtplib.py", line 309, in connect
    self.sock = self._get_socket(host, port, self.timeout)
  File "/usr/lib/python2.7/smtplib.py", line 284, in _get_socket
    return socket.create_connection((port, host), timeout)
  File "/usr/lib/python2.7/socket.py", line 553, in create_connection
    for res in getaddrinfo(host, port, 0, SOCK_STREAM):
gaierror: [Errno -2] Name or service not known

2015/12/20 16:46:28 - thermeq3 - DEBUG - Queqing [E]
2015/12/20 16:46:28 - thermeq3 - DEBUG - Message queue=['E']
2015/12/20 16:46:28 - thermeq3 - DEBUG - Sending message [E]
2015/12/20 16:46:29 - thermeq3 - ERROR - Error connecting to mail server mail.foo.local:25. Error code: [Errno -2] Name or service not known
2015/12/20 16:46:29 - thermeq3 - ERROR - Traceback: Traceback (most recent call last):
  File "/root/nsm.py", line 442, in sendEmail
    server = smtplib.SMTP(stp.mailserver, stp.mailport)
  File "/usr/lib/python2.7/smtplib.py", line 249, in __init__
    (code, msg) = self.connect(host, port)
  File "/usr/lib/python2.7/smtplib.py", line 309, in connect
    self.sock = self._get_socket(host, port, self.timeout)
  File "/usr/lib/python2.7/smtplib.py", line 284, in _get_socket
    return socket.create_connection((port, host), timeout)
  File "/usr/lib/python2.7/socket.py", line 553, in create_connection
[... 1253 lines cut ...]
Traceback (most recent call last):
  File "/root/nsm.py", line 1654, in <module>
    prepare()
  File "/root/nsm.py", line 1609, in prepare
    updateStatus("start")
NameError: global name 'updateStatus' is not defined
21:33:44 Redirection active
Traceback (most recent call last):
  File "/root/nsm.py", line 1654, in <module>
    prepare()
  File "/root/nsm.py", line 1609, in prepare
    updateStatus("start")
NameError: global name 'updateStatus' is not defined
21:43:44 Redirection active
Traceback (most recent call last):
  File "/root/nsm.py", line 1654, in <module>
    prepare()
  File "/root/nsm.py", line 1609, in prepare
    updateStatus("start")
NameError: global name 'updateStatus' is not defined
21:53:44 Redirection active
Traceback (most recent call last):
  File "/root/nsm.py", line 1654, in <module>
    prepare()
  Fi

autopower, 2015-12-20:
Your MAX!Cube refusing connection: "error: [Errno 146] Connection refused". Can you check Max!Cube IP and port for me? It's 192.168.0.222? [MAX! software](https://max.eq-3.de/login.jsp#download) is correctly communicating with MAX!Cube?

dominator99, 2015-12-20:
Hi

I can ping 195.168.0.222 successfully & wireless radstats seem to be 
talking to Max! cube

Which port should I be checking?

Regards

Chris

autopower, 2015-12-20:
Default port is 62910

dominator99, 2015-12-20:
I have ports Java Max 62910 & Java Max 23272 listed in Windows 7 
firewall as allowed (private & public)
Is that OK?

Regards

Chris

autopower, 2015-12-20:
Can you ping 192.168.0.222 from yun? And telnet to max!cube: `telnet 192.168.0.222 62910` from yun?

dominator99, 2015-12-20:
I can ping from Yun but telnet fails with:

root@ArduinoYun:~# telnet 192.168.0.222 62910
telnet: can't connect to remote host (192.168.0.222): Connection refused

I also tried from Windows 7 & got:

Microsoft Telnet> open 192.168.0.222 62910
Connecting To 192.168.0.222...Could not open connection to the host, on 
port 62910:  Connect failed

Any ideas?

Regards

Chris

autopower, 2015-12-20:
if you can correctly ping max!cube from win and from yun, there must be another port on max!cube in use, please check port 80 on 192.168.0.222.
Can you scan open ports on max!cube? For example with: http://www.advanced-port-scanner.com/
Or use your favourite port scanner.

dominator99, 2015-12-20:
Hi

see attached port scan results - does this mean port 22 is the only open 
port on the Max! cube wireless connection?

192.168.0.3 is the Max! cube wired connection

Regards

Chris

autopower, 2015-12-21:
Hello Chris,
just to be sure. Do you have these devices:
* [MAX!Cube](http://www.eq-3.de/max-heizungssteuerung-produktdetail-kopie/items/bc-lgw-o-tw.html) with address 192.168.0.222
* [Arduino Yun](https://www.arduino.cc/en/Main/ArduinoBoardYun) with address <192.168.0.???>
* [valves](http://www.eq-3.de/max-heizungssteuerung-produktdetail-kopie/items/bc-rt-trx-cyg.html) connected via radio to MAX!Cube
Please type yun address.

dominator99, 2015-12-23:
Hi Michal

After another Yun 'bricking' I finally got Yun working again, followed 
your readme.md instructions & voila it worked well for 5 minutes anyway!

On my bench simulation I have the loop (green) LED, the red (error) LED, 
the blue (status) LED & the yellow (relay) LED set up on a breadboard.
Initially the green LED was flashing (indicating system was running) 
then the blue & Yellow LED's lit (indicating the relay was ON) but after 
a few minutes the blue & yellow LED's  turned OFF & the red error LED lit.
I rebooted the Yun but same problem.

I've attached logs for your perusal

For the Max! cube I used the address: 192.168.0.3 in config.py (this is 
the only address reported by the port scanner but I would have expected 
2 IP addresses, 1 for wired & 1 for wireless like the Yun).

For the Yun the port scanner reported 2 IP addresses, 192.168.0.5 
(wired) & 192.168.0.12 (wireless)

2 steps forward 1 step back but going in the right direction :)

Regards

Chris

Lounge party wall Radstat,Lounge party wall Radstat,Radiator Thermostat 1,Radiator Thermostat 1,Lounge gable end Radstat,Lounge gable end Radstat,Radiator Thermostat 1,Radiator Thermostat 1,Radstat Lottie's room,Radstat Lottie's room,Radstat spare room,Radstat spare room,Bathroom Radstat,Bathroom Radstat,
22/12/2015 23:29:14,0,65,22.0,34,21.5,54,22.0,0,19.0,0,17.0,0,255,0,17.0,
22/12/2015 23:30:55,1,65,22.0,34,21.5,54,22.0,0,19.0,0,17.0,0,255,0,17.0,
22/12/2015 23:32:35,1,65,22.0,34,21.5,54,22.0,0,19.0,0,17.0,0,255,0,17.0,
22/12/2015 23:34:15,1,65,22.0,34,21.5,54,22.0,0,19.0,0,17.0,0,255,0,17.0,
22/12/2015 23:35:57,1,65,22.0,34,21.5,54,22.0,0,19.0,0,17.0,0,255,0,17.0,
22/12/2015 23:37:37,1,52,22.0,34,21.5,54,22.0,0,19.0,0,17.0,0,255,0,17.0,
22/12/2015 23:39:27,1,52,22.0,34,21.5,54,22.0,0,19.0,0,17.0,0,255,0,17.0,
22/12/2015 23:41:08,1,52,22.0,34,21.5,54,22.0,0,19.0,0,17.0,0,255,0,17.0,
22/12/2015 23:42:49,1,34,22.0,34,21.5,54,22.0,0,19.0,0,17.0,0,255,0,17.0,

2015/12/22 18:11:45 - thermeq3 - INFO - --> V147 started with PID=989 <--
2015/12/22 18:11:51 - thermeq3 - ERROR - Bridge error codeword [heattime] value [{'total': [0, 0.0], '22-12-2015': [0, 1450807905.3984958]}]
2015/12/22 23:28:52 - thermeq3 - ERROR - Error loading bridge file, using defaults!
2015/12/22 23:28:52 - thermeq3 - DEBUG - Queqing [S]
2015/12/22 23:28:52 - thermeq3 - DEBUG - Message queue=['S']
2015/12/22 23:28:52 - thermeq3 - DEBUG - Sending message [S]
2015/12/22 23:28:56 - thermeq3 - DEBUG - Creating new daily key: 22-12-2015
2015/12/22 23:28:56 - thermeq3 - DEBUG - Day reading warnings value=0
2015/12/22 23:29:01 - thermeq3 - DEBUG - Bridge file saved.
2015/12/22 23:29:01 - thermeq3 - INFO - Zero sized stderr log file, nothing'll be send
2015/12/22 23:29:02 - thermeq3 - ERROR - Unable to get latest version info - HTTPError = Not Found
2015/12/22 23:29:08 - thermeq3 - DEBUG - Bridge file saved.
2015/12/22 23:29:09 - thermeq3 - DEBUG - Public IP address: 86.6.59.107
2015/12/22 23:29:09 - thermeq3 - INFO - Current temperature in Bratislava is 3, humidity 100%
2015/12/22 23:29:09 - thermeq3 - DEBUG - OWW interval updated to [780, 2340, 2340]
2015/12/22 23:29:09 - thermeq3 - DEBUG - Valve ignore interval updated to 34
2015/12/22 23:29:14 - thermeq3 - DEBUG - defaults, 35% at 1 valve(s).
2015/12/22 23:29:14 - thermeq3 - DEBUG - Actual positions:
Room: Lottie's bedroom
	[124620] Radstat Lottie's room@  0% @ 17.0'C #  0.0'C (+)
Room: Bathroom
	[127360] Bathroom Radstat    @  0% @ 17.0'C #  0.0'C (+)
Room: Dining room
	[124286] Radiator Thermostat 1@ 34% @ 21.5'C #  0.0'C (+)
Room: Lounge gable end wall
	[127C1F] Lounge gable end Radstat@ 54% @ 22.0'C # 21.8'C (+)
Room: Spare room
	[124393] Radstat spare room  @  0% @  255'C #  255'C (+)
Room: Study
	[12433C] Radiator Thermostat 1@  0% @ 19.0'C #  0.0'C (+)
Room: Lounge party wall
	[127D37] Lounge party wall Radstat@ 65% @ 22.0'C # 21.4'C (+)
[... 454 lines cut ...]

status=idle
ignore_opened=34
svpnmw=75
interval=90
autoupdate=1
error=48
total_switch=150
uptime=0:12:22
current_status={"Lottie's bedroom": {'124620': ["Radstat Lottie's room", '0', '17.0', '0.0', '1']}, 'Bathroom': {'127360': ['Bathroom Radstat', '0', '17.0', '0.0', '1']}, 'Dining room': {'124286': ['Radiator Thermostat 1', '34', '21.5', '0.0', '1']}, 'Lounge gable end wall': {'127C1F': ['Lounge gable end Radstat', '54', '22.0', '21.8', '1']}, 'Spare room': {'124393': ['Radstat spare room', '0', '255', '255', '1']}, 'Study': {'12433C': ['Radiator Thermostat 1', '0', '19.0', '0.0', '1']}, 'Lounge party wall': {'127D37': ['Lounge party wall Radstat', '34', '22.0', '21.4', '1']}}
valve_pos=35
msg=
ignored={}
profile=time
heattime={'total': [816, 1450827778.6278901], '22-12-2015': [815, 1450827778.6290532]}
beta=no
preference=per
app_uptime=6:02:08
totalerrors=0
heattime_string=0:13:36
command=
daily=0:13:35
no_oww=0
mode=auto
valves=2

autopower, 2015-12-23:
Hello Chris,
this is just guess, find out these lines in nsm.py:
```
# on some system use code below:
# self.client_socket.setblocking(0)
```
and uncomment second one. Code will look like:
```
# on some system use code below:
self.client_socket.setblocking(0)
```

dominator99, 2015-12-23:
Hi Michal

Turned system on this morning & instead of the solid ref (error) LED I 
got the 'disco' lights.

Tried your suggested mod below but after reboot still get the disco lights.

Why did it appear to work for the first 5 minutes yesterday?

PS

In my nsm.py the lines read:

|# on some system use code below: # var.client_socket.setblocking(0) not ||# on some system use code below: # self.client_socket.setblocking(0) I 
changed mine to read: ||# on some system use code below: var.client_socket.setblocking(0) PPS I 
rebooted the Max! cube & the 'disco' lights have not returned (yet!) 
I'll report back later Many thanks again for getting me this far |

Regards

Chris

autopower, 2015-12-23:
Can you post or [mail](mailto:autopowerdevice@gmail.com) log file, error log (if any)?
Disco lights comes only if 'quit' command received.

dominator99, 2015-12-23:
Hi Michal

attached files as requested - there are no entries in the error log 
(take that as a good sign)

Regards

Chris

2015/12/22 18:11:45 - thermeq3 - INFO - --> V147 started with PID=989 <--
2015/12/22 18:11:51 - thermeq3 - ERROR - Bridge error codeword [heattime] value [{'total': [0, 0.0], '22-12-2015': [0, 1450807905.3984958]}]
2015/12/22 23:28:52 - thermeq3 - ERROR - Error loading bridge file, using defaults!
2015/12/22 23:28:52 - thermeq3 - DEBUG - Queqing [S]
2015/12/22 23:28:52 - thermeq3 - DEBUG - Message queue=['S']
2015/12/22 23:28:52 - thermeq3 - DEBUG - Sending message [S]
2015/12/22 23:28:56 - thermeq3 - DEBUG - Creating new daily key: 22-12-2015
2015/12/22 23:28:56 - thermeq3 - DEBUG - Day reading warnings value=0
2015/12/22 23:29:01 - thermeq3 - DEBUG - Bridge file saved.
2015/12/22 23:29:01 - thermeq3 - INFO - Zero sized stderr log file, nothing'll be send
2015/12/22 23:29:02 - thermeq3 - ERROR - Unable to get latest version info - HTTPError = Not Found
2015/12/22 23:29:08 - thermeq3 - DEBUG - Bridge file saved.
2015/12/22 23:29:09 - thermeq3 - DEBUG - Public IP address: 86.6.59.107
2015/12/22 23:29:09 - thermeq3 - INFO - Current temperature in Bratislava is 3, humidity 100%
2015/12/22 23:29:09 - thermeq3 - DEBUG - OWW interval updated to [780, 2340, 2340]
2015/12/22 23:29:09 - thermeq3 - DEBUG - Valve ignore interval updated to 34
2015/12/22 23:29:14 - thermeq3 - DEBUG - defaults, 35% at 1 valve(s).
2015/12/22 23:29:14 - thermeq3 - DEBUG - Actual positions:
Room: Lottie's bedroom
	[124620] Radstat Lottie's room@  0% @ 17.0'C #  0.0'C (+)
Room: Bathroom
	[127360] Bathroom Radstat    @  0% @ 17.0'C #  0.0'C (+)
Room: Dining room
	[124286] Radiator Thermostat 1@ 34% @ 21.5'C #  0.0'C (+)
Room: Lounge gable end wall
	[127C1F] Lounge gable end Radstat@ 54% @ 22.0'C # 21.8'C (+)
Room: Spare room
	[124393] Radstat spare room  @  0% @  255'C #  255'C (+)
Room: Study
	[12433C] Radiator Thermostat 1@  0% @ 19.0'C #  0.0'C (+)
Room: Lounge party wall
	[127D37] Lounge party wall Radstat@ 65% @ 22.0'C # 21.4'C (+)
2015/12/22 23:29:19 - thermeq3 - INFO - heating started due to room Lounge gable end wall, valve Lounge gable end Radstat@[54, 22.0, 21.8]
2015/12/22 23:29:20 - thermeq3 - DEBUG - Queqing [H]
2015/12/22 23:29:20 - thermeq3 - DEBUG - Message queue=['H']
2015/12/22 23:29:20 - thermeq3 - DEBUG - Sending message [H]
2015/12/22 23:30:55 - thermeq3 - DEBUG - heating, 35% at 2 valve(s).
2015/12/22 23:30:55 - thermeq3 - DEBUG - Actual positions:
Room: Lottie's bedroom
	[124620] Radstat Lottie's room@  0% @ 17.0'C #  0.0'C (+)
Room: Bathroom
	[127360] Bathroom Radstat    @  0% @ 17.0'C #  0.0'C (+)
Room: Dining room
	[124286] Radiator Thermostat 1@ 34% @ 21.5'C #  0.0'C (+)
Room: Lounge gable end wall
	[127C1F] Lounge gable end Radstat@ 54% @ 22.0'C # 21.8'C (+)
Room: Spare room
	[124393] Radstat spare room  @  0% @  255'C #  255'C (+)
Room: Study
	[12433C] Radiator Thermostat 1@  0% @ 19.0'C #  0.0'C (+)
Room: Lounge party wall
	[127D37] Lounge party wall Radstat@ 65% @ 22.0'C # 21.4'C (+)
2015/12/22 23:32:35 - thermeq3 - DEBUG - heating, 35% at 2 valve(s).
2015/12/22 23:32:35 - thermeq3 - DEBUG - Actual positions:
Room: Lottie's bedroom
	[124620] Radstat Lottie's room@  0% @ 17.0'C #  0.0'C (+)
Room: Bathroom
	[127360] Bathroom Radstat    @  0% @ 17.0'C #  0.0'C (+)
Room: Dining room
	[124286] Radiator Thermostat 1@ 34% @ 21.5'C #  0.0'C (+)
Room: Lounge gable end wall
[... 1154 lines cut ...]
	[127D37] Lounge party wall Radstat@  0% @ 20.0'C #  0.0'C (+)
2015/12/23 09:53:01 - thermeq3 - DEBUG - Bridge file saved.
2015/12/23 09:53:01 - thermeq3 - DEBUG - Public IP address: 86.6.59.107
2015/12/23 09:53:02 - thermeq3 - INFO - Difference between Yahoo and OWM temperatures. Yahoo=5.0 OWM=3.0
2015/12/23 09:53:02 - thermeq3 - INFO - Current temperature in Bratislava is 5, humidity 96%
2015/12/23 09:53:02 - thermeq3 - DEBUG - OWW interval updated to [840, 2520, 2520]
2015/12/23 09:53:03 - thermeq3 - DEBUG - Valve ignore interval updated to 38
2015/12/23 09:54:07 - thermeq3 - DEBUG - idle, 35% at 2 valve(s).
2015/12/23 09:54:07 - thermeq3 - DEBUG - Actual positions:
Room: Lottie's bedroom
	[124620] Radstat Lottie's room@  0% @ 17.0'C #  0.0'C (+)
Room: Bathroom
	[127360] Bathroom Radstat    @  0% @ 17.0'C #  0.0'C (+)
Room: Dining room
	[124286] Dining room radstat @  8% @ 20.0'C #  0.0'C (+)
Room: Lounge gable end wall
	[127C1F] Lounge gable end Radstat@  0% @ 20.0'C #  0.0'C (+)
Room: Spare room
	[124393] Spare room radstat  @  0% @  255'C #  255'C (+)
Room: Study
	[12433C] Study radstat       @  3% @ 19.0'C #  0.0'C (+)
Room: Lounge party wall
	[127D37] Lounge party wall Radstat@  0% @ 20.0'C #  0.0'C (+)
2015/12/23 09:55:48 - thermeq3 - DEBUG - idle, 35% at 2 valve(s).
2015/12/23 09:55:48 - thermeq3 - DEBUG - Actual position

dominator99, 2015-12-23:
Just after I sent the below I noticed the log file wasn't updating since 
I sent below.

Now when trying tu run WinSCP I get an error message:

Error changing directory to '/mnt/sda1'

Command 'cd "/mnt/sda1"'
failed with return code 0 and error message
-ash: groups: not found.

I assume this is the reason as sda1 (micro SD card) can't be mounted.

I tried rebooting Yun & Max! cube several times but still the same problem.

I also can't trigger the relay with the radstat 'boost' as before.

I assume all these issues are connected

Any suggestions?

regards

Chris

Hi Michal

attached files as requested - there are no entries in the error log 
(take that as a good sign)

Regards

Chris

2015/12/22 18:11:45 - thermeq3 - INFO - --> V147 started with PID=989 <--
2015/12/22 18:11:51 - thermeq3 - ERROR - Bridge error codeword [heattime] value [{'total': [0, 0.0], '22-12-2015': [0, 1450807905.3984958]}]
2015/12/22 23:28:52 - thermeq3 - ERROR - Error loading bridge file, using defaults!
2015/12/22 23:28:52 - thermeq3 - DEBUG - Queqing [S]
2015/12/22 23:28:52 - thermeq3 - DEBUG - Message queue=['S']
2015/12/22 23:28:52 - thermeq3 - DEBUG - Sending message [S]
2015/12/22 23:28:56 - thermeq3 - DEBUG - Creating new daily key: 22-12-2015
2015/12/22 23:28:56 - thermeq3 - DEBUG - Day reading warnings value=0
2015/12/22 23:29:01 - thermeq3 - DEBUG - Bridge file saved.
2015/12/22 23:29:01 - thermeq3 - INFO - Zero sized stderr log file, nothing'll be send
2015/12/22 23:29:02 - thermeq3 - ERROR - Unable to get latest version info - HTTPError = Not Found
2015/12/22 23:29:08 - thermeq3 - DEBUG - Bridge file saved.
2015/12/22 23:29:09 - thermeq3 - DEBUG - Public IP address: 86.6.59.107
2015/12/22 23:29:09 - thermeq3 - INFO - Current temperature in Bratislava is 3, humidity 100%
2015/12/22 23:29:09 - thermeq3 - DEBUG - OWW interval updated to [780, 2340, 2340]
2015/12/22 23:29:09 - thermeq3 - DEBUG - Valve ignore interval updated to 34
2015/12/22 23:29:14 - thermeq3 - DEBUG - defaults, 35% at 1 valve(s).
2015/12/22 23:29:14 - thermeq3 - DEBUG - Actual positions:
Room: Lottie's bedroom
	[124620] Radstat Lottie's room@  0% @ 17.0'C #  0.0'C (+)
Room: Bathroom
	[127360] Bathroom Radstat    @  0% @ 17.0'C #  0.0'C (+)
Room: Dining room
	[124286] Radiator Thermostat 1@ 34% @ 21.5'C #  0.0'C (+)
Room: Lounge gable end wall
	[127C1F] Lounge gable end Radstat@ 54% @ 22.0'C # 21.8'C (+)
Room: Spare room
	[124393] Radstat spare room  @  0% @  255'C #  255'C (+)
Room: Study
	[12433C] Radiator Thermostat 1@  0% @ 19.0'C #  0.0'C (+)
Room: Lounge party wall
	[127D37] Lounge party wall Radstat@ 65% @ 22.0'C # 21.4'C (+)
2015/12/22 23:29:19 - thermeq3 - INFO - heating started due to room Lounge gable end wall, valve Lounge gable end Radstat@[54, 22.0, 21.8]
2015/12/22 23:29:20 - thermeq3 - DEBUG - Queqing [H]
2015/12/22 23:29:20 - thermeq3 - DEBUG - Message queue=['H']
2015/12/22 23:29:20 - thermeq3 - DEBUG - Sending message [H]
[... 1167 lines cut ...]
	[124620] Radstat Lottie's room@  0% @ 17.0'C #  0.0'C (+)
Room: Bathroom
	[127360] Bathroom Radstat    @  0% @ 17.0'C #  0.0'C (+)
Room: Dining room
	[124286] Dining room radstat @  8% @ 20.0'C #  0.0'C (+)
Room: Lounge gable end wall
	[127C1F] Lounge gable end Radstat@  0% @ 20.0'C #  0.0'C (+)
Room: Spare room
	[124393] Spare room radstat  @  0% @  255'C #  255'C (+)
Room: Study
	[12433C] Study radstat       @  3% @ 19.0'C #  0.0'C (+)
Room: Lounge party wall
	[127D37] Lounge party wall Radstat@  0% @ 20.0'C #  0.0'C (+)
2015/12/23 09:53:01 - thermeq3 - DEBUG - Bridge file saved.
2015/12/23 09:53:01 - thermeq3 - DEBUG - Public IP address: 86.6.59.107
2015/12/23 09:53:02 - thermeq3 - INFO - Difference between Yahoo and OWM temperatures. Yahoo=5.0 OWM=3.0
2015/12/23 09:53:02 - thermeq3 - INFO - Current temperature in Bratislava is 5, humidity 96%
2015/12/23 09:53:02 - thermeq3 - DEBUG - OWW interval updated to [840, 2520, 2520]
2015/12/23 09:53:03 - thermeq3 - DEBUG - Valve ignore interval updated to 38
2015/12/23 09:54:07 - thermeq3 - DEBUG - idle, 35% at 2 valve(s).
2015/12/23 09:54:07 - thermeq3 - DEBUG - Actual positions:
Room: Lottie's bedroom
	[124620] Radstat Lottie's room@  0% @ 17.0'C #  0.0'C (+)
Room: Bathroom
	[127360] Bathro

dominator99, 2015-12-23:
Update

Also just noticed that the FAT32 & Ext4 partitions of the SD card don't 
have drive letters assigned when I check SD card using Aomei partition 
assistant under Windows 7.
Is this OK?

Just after I sent the below I noticed the log file wasn't updating since 
I sent below.

Now when trying tu run WinSCP I get an error message:

Error changing directory to '/mnt/sda1'

Command 'cd "/mnt/sda1"'
failed with return code 0 and error message
-ash: groups: not found.

I assume this is the reason as sda1 (micro SD card) can't be mounted.

I tried rebooting Yun & Max! cube several times but still the same problem.

I also can't trigger the relay with the radstat 'boost' as before.

I assume all these issues are connected

Any suggestions?

regards

Chris

Hi Michal

attached files as requested - there are no entries in the error log 
(take that as a good sign)

Regards

Chris

2015/12/22 18:11:45 - thermeq3 - INFO - --> V147 started with PID=989 <--
2015/12/22 18:11:51 - thermeq3 - ERROR - Bridge error codeword [heattime] value [{'total': [0, 0.0], '22-12-2015': [0, 1450807905.3984958]}]
2015/12/22 23:28:52 - thermeq3 - ERROR - Error loading bridge file, using defaults!
2015/12/22 23:28:52 - thermeq3 - DEBUG - Queqing [S]
2015/12/22 23:28:52 - thermeq3 - DEBUG - Message queue=['S']
2015/12/22 23:28:52 - thermeq3 - DEBUG - Sending message [S]
2015/12/22 23:28:56 - thermeq3 - DEBUG - Creating new daily key: 22-12-2015
2015/12/22 23:28:56 - thermeq3 - DEBUG - Day reading warnings value=0
2015/12/22 23:29:01 - thermeq3 - DEBUG - Bridge file saved.
2015/12/22 23:29:01 - thermeq3 - INFO - Zero sized stderr log file, nothing'll be send
2015/12/22 23:29:02 - thermeq3 - ERROR - Unable to get latest version info - HTTPError = Not Found
2015/12/22 23:29:08 - thermeq3 - DEBUG - Bridge file saved.
2015/12/22 23:29:09 - thermeq3 - DEBUG - Public IP address: 86.6.59.107
2015/12/22 23:29:09 - thermeq3 - INFO - Current temperature in Bratislava is 3, humidity 100%
2015/12/22 23:29:09 - thermeq3 - DEBUG - OWW interval updated to [780, 2340, 2340]
2015/12/22 23:29:09 - thermeq3 - DEBUG - Valve ignore interval updated to 34
2015/12/22 23:29:14 - thermeq3 - DEBUG - defaults, 35% at 1 valve(s).
2015/12/22 23:29:14 - thermeq3 - DEBUG - Actual positions:
Room: Lottie's bedroom
	[124620] Radstat Lottie's room@  0% @ 17.0'C #  0.0'C (+)
Room: Bathroom
	[127360] Bathroom Radstat    @  0% @ 17.0'C #  0.0'C (+)
Room: Dining room
	[124286] Radiator Thermostat 1@ 34% @ 21.5'C #  0.0'C (+)
Room: Lounge gable end wall
	[127C1F] Lounge gable end Radstat@ 54% @ 22.0'C # 21.8'C (+)
Room: Spare room
	[124393] Radstat spare room  @  0% @  255'C #  255'C (+)
Room: Study
[... 1169 lines cut ...]
Room: Lounge party wall
	[127D37] Lounge party wall Radstat@  0% @ 20.0'C #  0.0'C (+)
2015/12/23 09:52:18 - thermeq3 - DEBUG - idle, 35% at 2 valve(s).
2015/12/23 09:52:18 - thermeq3 - DEBUG - Actual positions:
Room: Lottie's bedroom
	[124620] Radstat Lottie's room@  0% @ 17.0'C #  0.0'C (+)
Room: Bathroom
	[127360] Bathroom Radstat    @  0% @ 17.0'C #  0.0'C (+)
Room: Dining room
	[124286] Dining room radstat @  8% @ 20.0'C #  0.0'C (+)
Room: Lounge gable end wall
	[127C1F] Lounge gable end Radstat@  0% @ 20.0'C #  0.0'C (+)
Room: Spare room
	[124393] Spare room radstat  @  0% @  255'C #  255'C (+)
Room: Study
	[12433C] Study radstat       @  3% @ 19.0'C #  0.0'C (+)
Room: Lounge party wall
	[127D37] Lounge party wall Radstat@  0% @ 20.0'C #  0.0'C (+)
2015/12/23 09:53:01 - thermeq3 - DEBUG - Bridge file saved.
2015/12/23 09:53:01 - thermeq3 - DEBUG - Public IP address: 86.6.59.107
2015/12/23 09:53:02 - thermeq3 - INFO - Difference between Yahoo and OWM temperatures. Yahoo=5.0 OWM=3.0
2015/12/23 09:53:02 - thermeq3 - INFO - Current temperature in Bratislava is 5, humidity 96%
2015/12/23 09:53:02 - thermeq3 - DEBUG - OWW interval updated to [840, 2520, 2520]
2015/12/23 09:53:03 - thermeq3 - DEBUG - Valve ignore interval updated to 38
2015/12/23 09:54:07 - thermeq3 - DEBUG - idle,

autopower, 2015-12-23:
I'm little bit confused with your setup, can you open ssh port for me? Please change password to default, so I can check whats going on.

autopower, 2015-12-23:
Check out [diag.sh](https://github.com/autopower/thermeq3/blob/master/install/diag/diag.sh)

dominator99, 2015-12-28:
Hi Michal

sorry about delay but my Yun has 'bricked' permanently this time & I 
can't reset to factory settings.
I've ordered a new one & will get back to you when up & running again

Regards

Chris

dominator99, 2016-01-10:
Hi Michal

I've had the thermeq3 running on the bench successfully for the last few 
days & have a couple of questions.

1.    How do I get a full days set of readings from the csv file (like 
hellmostat_20160110-000822.csv) for graphing     purposes?  The csv 
files seems to be created randomly.  The above file has about 20 minutes 
of readings &         hellmostat_20160110-115032.csv has no readings at 
all after approx6 hours running time!

2.    What is the meaning of the value after the # sign in:
         [124286] Dining room stat    @  0% @ 20.0'C # _/21.5'C (+)/_

-- 

Regards

Chris

autopower, 2016-01-10:
Hi Chris,
ad 1) file you mentioned is from 10/jan/2010, 00:08:22. This file will be closed at midnight. Second one (10/jan/2016 = 20160110) was created 11:50:32. Files are created at midnight or after restart, so if you restarted app (or app crashed) during the day, new file is created. Check for other file from same day. And if there are any of them, please check error file and/or errors in log.
ad 2) # sign followed by value is current temperature, as read from valve. Sometimes is 0, because of design of eq3 system.

m

dominator99, 2016-01-10:
Thanks Michal

Regards

Chris

autopower, 2016-01-10:
You are welcome.

dominator99, 2016-01-14:
Hi Michal

All seems to be working normally (finally!)

Have you given any thought to creating graphs from the csv files? [I find graphs much easier to read than csv files :) ]
Would Google Charts lend itself to this?

dominator99, 2016-01-15:
Hi Michal

system did not shut down tonight as normal (arround 00:08:00) but continued to run heating.
I rebooted Yun but same behaviour; is this normal or because of low outside temps here (approx 2C)?

thanks in anticipation

autopower, 2016-01-15:
Hello Chris,
would you be so kind and attach log?

dominator99, 2016-01-15:
Hi Michal

see log attached as requested.  Any thoughts on the graphing of csv data

Regards

Chris

2016/01/12 00:00:02 - thermeq3 - DEBUG - heating, 35% at 2 valve(s).
2016/01/12 00:00:02 - thermeq3 - DEBUG - Actual positions:
Room: Lottie's bedroom
	[124620] Lottie's stat       @ 24% @ 17.0'C #  0.0'C (+)
Room: Bathroom
	[127360] Bathroom stat       @  0% @ 17.0'C #  0.0'C (+)
Room: Dining room
	[124286] Dining room stat    @ 34% @ 21.5'C # 21.7'C (+)
Room: Lounge gable end wall
	[127C1F] Gable end stat      @ 90% @ 23.0'C #  0.0'C (+)
Room: Spare room
	[124393] Spare room stat     @ 34% @ 18.0'C # 18.1'C (+)
Room: Study
	[12433C] Study stat          @ 35% @ 20.0'C # 24.7'C (+)
Room: Lounge party wall
	[127D37] Party wall stat     @100% @ 23.0'C # 25.2'C (+)
Room: Landing
	[1171C4] Landing stat        @ 61% @ 21.5'C # 21.1'C (+)
2016/01/12 00:01:42 - thermeq3 - DEBUG - heating, 35% at 2 valve(s).
2016/01/12 00:01:42 - thermeq3 - DEBUG - Actual positions:
Room: Lottie's bedroom
	[124620] Lottie's stat       @ 24% @ 17.0'C #  0.0'C (+)
Room: Bathroom
	[127360] Bathroom stat       @  0% @ 17.0'C #  0.0'C (+)
Room: Dining room
	[124286] Dining room stat    @ 34% @ 21.5'C # 21.7'C (+)
Room: Lounge gable end wall
	[127C1F] Gable end stat      @ 66% @ 22.0'C #  0.0'C (+)
Room: Spare room
	[124393] Spare room stat     @ 34% @ 18.0'C # 18.1'C (+)
Room: Study
	[12433C] Study stat          @ 35% @ 20.0'C # 24.7'C (+)
Room: Lounge party wall
	[127D37] Party wall stat     @100% @ 22.0'C # 25.2'C (+)
Room: Landing
	[1171C4] Landing stat        @ 73% @ 22.0'C # 21.1'C (+)
2016/01/12 00:03:21 - thermeq3 - DEBUG - heating, 35% at 2 valve(s).
2016/01/12 00:03:21 - thermeq3 - DEBUG - Actual positions:
Room: Lottie's bedroom
	[124620] Lottie's stat       @ 24% @ 17.0'C #  0.0'C (+)
Room: Bathroom
	[127360] Bathroom stat       @  0% @ 17.0'C #  0.0'C (+)
Room: Dining room
	[124286] Dining room stat    @ 54% @ 22.0'C # 21.7'C (+)
Room: Lounge gable end wall
	[127C1F] Gable end stat      @ 66% @ 22.0'C #  0.0'C (+)
Room: Spare room
	[124393] Spare room stat     @ 34% @ 18.0'C # 18.1'C (+)
Room: Study
	[12433C] Study stat          @ 47% @ 20.5'C # 24.7'C (+)
Room: Lounge party wall
	[127D37] Party wall stat     @100% @ 22.0'C # 25.2'C (+)
Room: Landing
	[1171C4] Landing stat        @ 73% @ 22.0'C # 21.1'C (+)
2016/01/12 00:05:01 - thermeq3 - DEBUG - heating, 35% at 2 valve(s).
2016/01/12 00:05:01 - thermeq3 - DEBUG - Actual positions:
Room: Lottie's bedroom
	[124620] Lottie's stat       @ 24% @ 17.0'C #  0.0'C (+)
Room: Bathroom
	[127360] Bathroom stat       @  0% @ 17.0'C #  0.0'C (+)
Room: Dining room
	[124286] Dining room stat    @ 54% @ 22.0'C # 21.7'C (+)
[... 1350 lines cut ...]
2016/01/12 12:35:15 - thermeq3 - DEBUG - Public IP address: 86.6.59.107
2016/01/12 12:35:15 - thermeq3 - INFO - Difference between Yahoo and OWM temperatures. Yahoo=5.0 OWM=3.0
2016/01/12 12:35:15 - thermeq3 - INFO - Current temperature in Bristol is 5, humidity 87%
2016/01/12 12:35:15 - thermeq3 - DEBUG - OWW interval updated to [840, 2520, 2520]
2016/01/12 12:35:16 - thermeq3 - DEBUG - Valve ignore interval updated to 38
2016/01/12 12:35:20 - thermeq3 - DEBUG - idle, 35% at 2 valve(s).
2016/01/12 12:35:20 - thermeq3 - DEBUG - Actual positions:
Room: Lottie's bedroom
	[124620] Lottie's stat       @  0% @ 17.0'C # 18.4'C (+)
Room: Bathroom
	[127360] Bathroom stat       @  0% @ 17.0'C #  0.0'C (+)
Room: Dining room
	[124286] Dining room stat    @  1% @ 20.0'C # 21.3'C (+)
Room: Lounge gable end wall
	[127C1F] Gable end stat      @  0% @ 21.5'C # 24.7'C (+)
Room: Spare room
	[124393] Spare room stat     @ 33% @ 18.0'C # 17.8'C (+)
Room: Study
	[12433C] Study stat          @  0% @ 18.5'C #  0.0'C (+)
Room: Lounge party wall
	[127D37] Party wall stat     @ 47% @ 21.5'C # 22.8'C (+)
Room: Landing
	[1171C4] Landing stat        @ 27% @ 20.0'C #  0.0'C (+)
2016/01/12 12:37:00 - thermeq3 - DEBUG - idle, 35% at 2 valve(s).
2016/01/12 12:37:00 - thermeq3 - DEBUG - Ac

dominator99, 2016-01-15:
above log seems to be truncated to 12Jan 2016 whereas my sent attachment shows up to 15 Jan 2016; is there a limit to the log file size on Github?

dominator99, 2016-01-15:
here's a truncated version of above log

[truncated hellmostat log.txt](https://github.com/autopower/thermeq3/files/91782/truncated.hellmostat.log.txt)

dominator99, 2016-01-16:
Hi Michal

Would it be possible to add the actual room temperatures read from the EQ-3 radstats to the daily csv files? (the temp after the # sign, I believe, in the single temp example below)
Room: Bathroom
	[127360] Bathroom stat       @  0% @ 17.0'C # 20.5'C (+)

autopower, 2016-01-17:
It's already there, first value is valve open [%], second one actual temperature [deg C]. Second one is 0 if thermostat reported so.

dominator99, 2016-01-17:
Surely the entry
 Room: Bathroom
[127360] Bathroom stat @ 0% @ 17.0'C # 20.5'C (+)
is from the daily log file not daily csv file.

My daily csv file has entries like 
Lounge party wall Radstat	Lounge party wall Radstat setting	Dining room radstat	Dining room radstat setting
but not the third log file entry shown above
Or am I missing something here :)

I don't understand your comment 
"It's already there, first value is valve open [%], second one actual temperature [deg C]. Second one is 0 if thermostat reported so."
In my daily csv files the second entry above only changes when the daily temperature program of the radstat valve changes & this is no third entry
Your comment
"Second one is 0 if thermostat reported so" has me puzzled :)
Should this read "Third one is 0 if thermostat reported so"?
I thought this third entry was the previous radstat temp reading?

dominator99, 2016-01-17:
I should have mentioned that I've managed to 'chart' the daily csv readings after opening with OpenOffice spreadsheet, which displays the radstat % valve opening quite nicely.  I would like to be able to include the varying daily room temps alongside the % valve opening hence for the request for including the room temps in the daily csv file where it can be imported into the spreadsheet easily.
I can see the room temps are included in the daily log but I can't import into the spreadsheet from there.
I hope the above explaination is clear :)

Thanks Michal (sorry to be a pain!)

dominator99, 2016-01-17:
Hi Michal

see attached copy of csv file & chart from Open Office spreadsheet

Regards

Chris

autopower, 2016-01-17:
At first, you are not a pain. Keep asking, any question can help another user.
ad csv: may be I'm lost but I didn't see any attachment. If your CSV looks like mine:
'17/01/2016 00:04:57,0,15,20.0,62,20.0,18,20.0,17,20.0,18,20.5,0,20.5,0,20.5'
it means: that 17/jan/2016 at 00:04:57 is heating off (0), first room valve is 15% opened, thermostat reading tells me that in room is 20.0 degC, second room valse is 62% opened, 20.0 degC, third 18% opened, 20.0 degC and so on. Just paste one line from CSV and I'll take a look...

dominator99, 2016-01-17:
here's 1st line of my csv

	0 = OFF 1 = ON 	% radstat valve open 	temp °C 	% radstat valve open 
temp °C 	% radstat valve open 	temp °C 	% radstat valve open 	temp °C 	% 
radstat valve open 	temp °C 	% radstat valve open 	temp °C 	% radstat 
valve open 	temp °C 	% radstat valve open 	temp °C
10/01/16 12:24 	0 	52 	21.5 	5 	20 	23 	18 	0 	20.5 	0 	17 	0 
21.5 	0 	17 	29 	20

The only time my 'temp C' box changes value is at the same time my 
room's programmed temps change, so it looks to me as though this 'temp 
C' box is not the actual room temp, which would change over time, but 
the programmed temp.

My first 'temp C' box reads 21.5 C from start of csv file @ 12:24 until 
16:22 when it changes to 22 C & stays at 22 C until 23:59 these are the 
pre programmed temp settings for this room NOT the actual room temps 
(which I would expect to fluctuate over time)

I hope this makes sense :)

Regards

Chris

dominator99, 2016-01-17:
here's 1st line of my csv

	0 = OFF 1 = ON 	% radstat valve open 	temp °C 	% radstat valve open 
temp °C 	% radstat valve open 	temp °C 	% radstat valve open 	temp °C 	% 
radstat valve open 	temp °C 	% radstat valve open 	temp °C 	% radstat 
valve open 	temp °C 	% radstat valve open 	temp °C
10/01/16 12:24 	0 	52 	21.5 	5 	20 	23 	18 	0 	20.5 	0 	17 	0 
21.5 	0 	17 	29 	20

The only time my 'temp C' box changes value is at the same time my 
room's programmed temps change, so it looks to me as though this 'temp 
C' box is not the actual room temp, which would change over time, but 
the programmed temp.

My first 'temp C' box reads 21.5 C from start of csv file @ 12:24 until 
16:22 when it changes to 22 C & stays at 22 C until 23:59 these are the 
pre programmed temp settings for this room NOT the actual room temps 
(which I would expect to fluctuate over time)

I hope this makes sense :)

ps

this is copied from my hellmostat log file

2016/01/17 23:51:45 - thermeq3 - DEBUG - heating, 35% at 2 valve(s).
2016/01/17 23:51:45 - thermeq3 - DEBUG - Actual positions:
Room: Lottie's bedroom
     [124620] Lottie's stat       @ 47% @ 17.0'C # 16.0'C (+)
Room: Bathroom
     [127360] Bathroom stat       @  0% @ 17.0'C # 20.7'C (+)
Room: Dining room
     [124286] Dining room stat    @100% @ 21.5'C # 21.4'C (+)
Room: Lounge gable end wall
     [127C1F] Gable end stat      @100% @ 22.5'C # 20.8'C (+)
Room: Spare room
     [124393] Spare room stat     @ 44% @ 18.0'C # 20.6'C (+)
Room: Study
     [12433C] Study stat          @ 48% @ 19.0'C # 20.0'C (+)
Room: Lounge party wall
     [127D37] Party wall stat     @100% @ 22.5'C # 21.5'C (+)
Room: Landing
     [1171C4] Landing stat        @ 80% @ 21.5'C # 21.9'C (+)

The temp entry after the valve open position is the radstat programmed 
temp not the actual room temp (which I think is the last entry)

Regards

Chris

dominator99, 2016-01-19:
Hi Michal

just found this:

      Actual Temperature (HeaterThermostat)

If a HeaterThermostat is in 'auto' mode, the actual temperature is 
sometimes returned at offset 8-9. This seems to only appear when the 
valve moves, although changing the set temperature might be enough. In 
adition, if the HeaterThermostat is in a room that has a linked 
WallMountedThermostat, this shows the temperature measured by the 
WallMountedThermostat instead.

|8 Actual Temperature 2 205 offset| 8 | ... | 9 | hex | 01 | | 32 | 
binary| 0000 0001 | ... | 0011 0010 | | |||| |||| +---------++++-++++--- 
actual temperature (°C*10): 100110010 = 30.6°C Would this explain why 
the last figure in: |

Room: Bathroom
     [127360] Bathroom stat       @  0% @ 17.5'C #  0.0'C (+) for 
example, is 0.0?

This figure of 0.0 will only change when the radstat valve changes 
position (moves)?

Regards

Chris

dominator99, 2016-01-20:
|Hi Michal On a couple of occasions the green (activity) LED stopped 
flashing as normal & remained lit constantly. The only way to resolve 
this was a hard reset. Does the Yun (thermeq3 sketch) support watchdog 
timer (WDT) reset? I found this online; would it work if embedded in 
your thermeq3 sketch? TIA #include <avr/wdt.h> void setup () {   
Serial.begin (9600);   Serial.println ("Restarted.");   wdt_enable 
(WDTO_1S);  // reset after one second, if no "pat the dog" received }  
// end of setup void loop () {   Serial.println ("Entered loop ...");   
wdt_reset ();  // give me another second to do stuff (pat the dog)   
while (true) ;   // oops, went into a loop }  // end of loop |

-- 

Regards

Chris

autopower, 2016-01-21:
Please check latest sketch (ino file). It's a LED issue, not leonardo part freezing.

dominator99, 2016-01-23:
Hi Michal

Any progress on the actual room temps (not the valve temp setting) being 
included in daily csv file

Regards

Chris

dominator99, 2016-02-15:
Hi Michal

Sorry to bother you but have you looked into including the actual room temps ..... # 21.9'C (+)
(not the valve temp setting) being included in daily csv file ?

Regards

Chris

autopower, 2016-02-16:
It's already there, V148, check for "uncomment" word in code and uncomment what you need

dominator99, 2016-02-16:
Hi Michal
Updated as above & got a csv file that showed the actual room temps but not for long.
The system no longer seems to be updating after 11:03 - see attached

[image]

I've tried rebooting the Yun; reinstalled your software; undone the uncomment; rebooted the Max! cube, still the same problem.

I noticed that the 3 attached text files are date stamped at 03/10/15 @ 13:54 but contain entries for today (16/02/2016) ??

[hellmostat_bridge.txt](https://github.com/autopower/thermeq3/files/132739/hellmostat_bridge.txt)
[hellmostat_log.txt](https://github.com/autopower/thermeq3/files/132741/hellmostat_log.txt)
[hellmostat_log_2015-09-08.txt](https://github.com/autopower/thermeq3/files/132740/hellmostat_log_2015-09-08.txt)

Any suggestions?

regards

chris

dominator99, 2016-03-14:
Hi Michal

I noticed from your facebook page that you recommend updating thermeq3 
to v2xx but I can't find any reference to it on Github, only the 
original V152

"In order to upgrade thermeq3 to 2xx version please convert your bridge 
file with: https://github.com/…/ma…/upgrade%20support/convert_bridge.py
And copy it to the /root directory of yún."

Can you explain this in more detail as I don't understand

-- 

Regards

Chris

dominator99, 2016-03-15:
I think I managed to install thermeq3 v2xx but the old problem of not showing actual room temps has returned 
In lines 1159 to 1163 of msn.py shown below I noticed that the 'commented' & 'uncommented' lines are the same!

        rooms[room_id][0] = roomStr
        # comment line below to use current temp
        var.csv.write(str(v[0]) + "," + str(v[1]) + ",")
        # uncomment line below to use current temp
        # var.csv.write(str(v[0]) + "," + str(v[1]) + ",")

My hellmostat log file contains entries like:

Room: Lottie's bedroom
	[124620] Lottie's stat       @ 80% @ 18.0'C #  0.0'C (+)
Room: Bathroom
	[127360] Bathroom stat       @  0% @ 17.0'C #  0.0'C (+)
Room: Dining room
	[124286] Dining room stat    @100% @ 20.5'C #  0.0'C (+)
Room: Lounge gable end wall
	[127C1F] Gable end stat      @100% @ 21.5'C #  0.0'C (+)
Room: Spare room
	[124393] Spare room stat     @ 34% @ 17.0'C #  0.0'C (+)
Room: Study
	[100831] Study radstat       @ 45% @ 18.5'C #  0.0'C (+)
Room: Lounge party wall
	[127D37] Party wall stat     @100% @ 21.5'C #  0.0'C (+)
Room: Landing
	[1171C4] Landing stat        @ 14% @ 20.0'C #  0.0'C (+)

How can I get the last temp 0.0'C to show the actual room temp recorded by the thermostat?

autopower, 2016-03-15:
As I see github version of nsm.py, 3 days old: 
`
        # comment line below to use current temp 
        var.csv.write(str(v[0]) + "," + str(v[1]) + ",")
        # uncomment line below to use current temp
        # var.csv.write(str(v[0]) + "," + str(v[2]) + ",")
`

dominator99, 2016-03-15:
HI Michal

using this setup:

         rooms[room_id][0] = roomStr
         # comment line below to use current temp
         # var.csv.write(str(v[0]) + "," + str(v[1]) + ",")
         # uncomment line below to use current temp
         var.csv.write(str(v[0]) + "," + str(v[2]) + ",")

produces this result:

Room: Lottie's bedroom
     [124620] Lottie's stat       @ 70% @ 19.0'C #  0.0'C (+)
Room: Bathroom
     [127360] Bathroom stat       @  0% @ 17.0'C #  0.0'C (+)
Room: Dining room
     [124286] Dining room stat    @ 80% @ 20.5'C #  0.0'C (+)
Room: Lounge gable end wall
     [127C1F] Gable end stat      @ 59% @ 21.5'C #  0.0'C (+)
Room: Spare room
     [124393] Spare room stat     @ 34% @ 17.0'C #  0.0'C (+)
Room: Study
     [100831] Study radstat       @ 45% @ 18.5'C #  0.0'C (+)
Room: Lounge party wall
     [127D37] Party wall stat     @ 86% @ 21.5'C #  0.0'C (+)
Room: Landing
     [1171C4] Landing stat        @ 15% @ 20.0'C #  0.0'C (+)

Still no final temp for room after #?

Regards

Chris

[11 more replies]