autopower/thermeq3 #7: Temperature above 25.6 degrees

FastFrank, 2016-07-14:
Last week the temperature outside was higher.
At some point (above 25.6 degrees) my heater started heating.
My wall thermostat was set at 69 degrees according to the log.
But is was set at 5.

I also found this on github:

https://github.com/Bouni/max-cube-protocol/blob/master/L-Message.md

On the page they say:

While valves have 11 byte long submessages a WallMountedThermostat adds another byte, making a 12 byte message.

In thermeq3 it's dev_len 12 for heater and dev_len 13 for wall thermostats.
I changed the code according (i tried) and now i get the (right) temperature from the thermostats all the time!
Except for those connected to the same room where the wall thermostat is in, they return zero.

```
if dev_len == 12:		# Heater thermostat
	valve_set = ord(es[es_pos + 0x08]) / 2.0
else: 					# Wall thermostat
	valve_set = (ord(es[es_pos + 0x08]) & 127) / 2.0
									
if valve_info & 3 != 2: # no vacation?
	if dev_len == 12: 	# Heater thermostat
		valve_temp = ord(es[es_pos + 0x0A]) / 10.0  + ((ord(es[es_pos + 0x09])&1)*256 ) / 10.0
	else:				# Wall thermostat
		valve_temp = (ord(es[es_pos + 0x0C]) + (ord(es[es_pos + 0x08]) & 128 / 128) * 256 ) / 10.0

```

FastFrank, 2016-07-20:
Maybe

`
valve_temp = (ord(es[es_pos + 0x0C]) + (ord(es[es_pos + 0x08]) & 128 ) * 2 ) / 10.0
`

works better...

autopower, 2016-07-20:
Hello,
which version? 1xx or 2xx?

FastFrank, 2016-07-21:
A version 1xx variant.

autopower, 2016-08-18:
Please check code in 2xx version.
e.g. 
lsb = ord(es[es_pos + 0x08]) & 0b01111111
msb = ((ord(es[es_pos + 0x08]) & 0b10000000) >> 7) * 256
 # get set temp
valve_temp = float(lsb) / 2
# get measured temp
 lsb = ord(es[es_pos + 0x0C])
 valve_curr_temp = float(msb + lsb) / 10
 # extract room name from this WallMountedThermostat
wall_room_id = str(self.devices[valve_adr][3])
# and update its value to current temperature as read from WallMountedthermostat
self.rooms[wall_room_id][3] = valve_curr_temp

FastFrank, 2016-10-29:
Sorry,
I noticed the code divides with integer values: 2 and 10
So the result is an integer too.

Python 2.7.3 (default, Aug  8 2013, 22:36:42)
[GCC 4.6.4 20121210 (prerelease)] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> 11 / 2
5
>>> 11 / 2.0
5.5

Do you get only whole numbers for valve_temp and valve_curr_temp?

stephenmhall, 2016-11-06:
Hi guys I have a RPi / Python MAX Boiler controller and had the same problem. Stems from the fact that the temp is split over bytes at both ends of the message, so when it goes over 25.6 it flips the lower byte to 1 and goes back to 00 on the higher bits I solved it by building a binary text string and converting it back.

`
Get Binary at offset 8
valve_binary = "{0:08b}".format(int(self.hexify(es[es_pos + 0x08]), 16))

Use msb of offset 8 for msb of actual temperature
if valve_binary[0:1] == "1":
    temp_headder = "01"
else:
    temp_headder = "00"

Just use last 6 bits for set temp
valve_binary = valve_binary[-6:]
valve_temp = float(int(valve_binary, 2)) / 2  # set temp

curr_binary = "{0}{1:08b}".format(temp_headder,int(self.hexify(es[es_pos + 0x0C]), 16))
valve_curtemp = float(int(curr_binary, 2)) / 10  # measured temp

Hope that is of some use.
`

autopower, 2016-11-13:
ad divide with int value of 2: there is a float(x) / 2 so the result will be float
ad msn/lsb: https://github.com/Bouni/max-cube-protocol
I'll check the code for wall mounted and HT. MSB is on bit 7 and bit 0 (HT).

closed 2016-12-04 by autopower