autopower/thermeq3 #8: disable the duty circle.

chrisj8727, 2016-09-07:
it is possible to disable the duty circle.
This is annoying especially during testing.
thanks in advance chris

autopower, 2016-09-21:
What you mean by duty circle?

chrisj8727, 2016-09-21:
The duty cycle describes a statutory limit of the airtime of devices in the
868 MHz range. The objective of this scheme is to the function of all in
the 868 MHz range to ensure working devices. In the frequency range used by
us, the maximum transmission time of each unit 868 MHz is 1% of one hour
(so 36 seconds in an hour). The devices may be used in achieving the
1%-limit no longer send, until the time limit is over.

chris

autopower, 2016-09-21:
Thermeq3 does not manipulate communication between device and cube in any way, so informatiom stored "in cube" is based on normal transmission between device and cube.

closed 2016-11-13 by autopower