autopower/thermeq3 #11: Antifreeze safety feature

bilbolodz, 2016-11-16:
I think it could be a good idea to implement "antifreeze safety feature":

add the temperature sensor to arduino and if all goes TOTALLY wrong (e.x. no connection to EQ-3 MAX! cube for long time) and temperature drops bellow threshold turn the boiler on to prevent possibility of freezing (maybe there is only network error or cube is not working but radiator thermostat can turn valve on but it needs a hot water).
What do you think about it?

autopower, 2016-11-16:
As far I know almost every boiler is equipped with antifreeze function. Despite that, if you are interested, I can add such a functionality into the sketch (e.g. DHT11 library and some code to get value from sensor and fire boiler).

bilbolodz, 2016-11-16:
I think that it could be a good safety feature. Question is: DHT11 or 1wire DS1820 (it could be more that one).

And second issue:
Do you have any experience with circulation pump for heating? My is (probably) controlled from boiler, I've to check but I'm afraid that my pump is running all the time......

To make your device "complete" it should be one more relay (for driving circulation pump) and one more sensor for measuring water temperature (it could be a little tricky but maybe it's enough to put sensor at the pipe and measure water temperature "not directly").
Water temperature sensor should ask boiler "for heating" and valves position should ask for "water circulation".
What do you think about such solution?

autopower, 2016-11-16:
ad temperature) I can write down some code with DHT11 and then with 1wire
ad pump) this is a good point

bilbolodz, 2016-11-16:
I'm waiting for my Arduino Yun module. When I get it I can write a code for Arduino.

bilbolodz, 2016-11-19:
I've checked more thoroughly my installation. It turned out that current thermostat is turning on/off circulation pump and boiler itself has own temperature regulator on the front.

bilbolodz, 2016-12-05:
I wrote a new version of sketch with a few additional features:
- antifreeze feature (require DHT22 sensor)
- possibility to use popular optocoupler driven relay board (active LOW not HIGH, you can bay it on ebay at very low price)
- possibility to power reset cube (require second relay connected to power supply of Cube and adding "r" command in python code - python code NOT implemented yet).

If you find it useful you can merge it into your project: https://github.com/bilbolodz/thermeq3 (and implement command "r" for reseting cube. Sketch was NOT tested yet in hardware I will do it tomorrow.

In my setup I've implemented very cheap and easy way of powering Cube over ethernet wire from "central UPS". In my case arduino, UPS and other network stuff is located at 19 inch rack "in basement" but Cube seats and center of house to provide good RF communication. Both elements (Cube and arduino) are powered from one power supply connected to UPS and these solution coste me less than 5 USD!

bilbolodz, 2016-12-05:
Have you installed library from http://playground.arduino.cc/Main/DHTLib (https://github.com/RobTillaart/Arduino/tree/master/libraries/DHTlib)?

bilbolodz, 2016-12-05:
It could be a problem. I've clean (completely fresh) install arduino-1.6.13 (linux version) and I only added these library.

closed 2016-12-12 by bilbolodz