autopower/thermeq3 #12: Beta "temp" mode not working

bilbolodz, 2016-12-02:
I'm trying to use (beta) "temp" mode after enabling it (http://XXX/data/put/profile/temp, http://XXX/data/put/beta/yes), I'm receiving an error:

`Traceback (most recent call last):
  File "/root/thermeq3/nsm.py", line 45, in <module>
    if t3.intervals() == 0xFF:
  File "/root/thermeq3/thermeq3.py", line 593, in intervals
    self._do_beta()
  File "/root/thermeq3/thermeq3.py", line 702, in _do_beta
    self._do_profiles()
  File "/root/thermeq3/thermeq3.py", line 688, in _do_profiles
    sm, am, kk = profiles.do(self.setup.selected_mode, self.var.act_mode_idx, self.var.situation)
  File "/root/thermeq3/profiles.py", line 107, in do
    kv = temp_mode()
  File "/root/thermeq3/profiles.py", line 76, in temp_mode
    c_temp = int(table.sit["current_temp"])
TypeError: int() argument must be a string or a number, not 'NoneType'
`

autopower, 2016-12-03:
Please check & test V224

bilbolodz, 2016-12-03:
Where (and when) will it be available?

autopower, 2016-12-03:
right now, in beta
https://github.com/autopower/thermeq3/tree/master/install/beta

bilbolodz, 2016-12-03:
Hmmm... How to do upgrade in "gentle way"?
I can see in log:
---
2016/12/03 21:25:33 [INFO] Available file: thermeq3.zip, V224 with hash 8ed1b49f3468f83bb1f30d401df2c613
2016/12/03 21:25:33 [INFO] Actual version: 223, hash: 8ed1b49f3468f83bb1f30d401df2c613
2016/12/03 21:25:33 [DEBUG] Update is not necessary.
---
???

autopower, 2016-12-03:
you can replace files in thermeq3 directory with V224
and then kill nsm.py process, or reset device

bilbolodz, 2016-12-03:
OK, I know, it's "usual way" :-) I've asked how to use your "auto update procedure" but OK I will replace it manually.

autopower, 2016-12-03:
just now I'm debugging autoupdate :)

bilbolodz, 2016-12-03:
So please check also this (after upgrade to 224) I can see in log:
`2016/12/03 21:39:59 [DEBUG] Message queue: ['H', 'S']
2016/12/03 21:39:59 [DEBUG] ## process_msg ## bridge:
2016/12/03 21:39:59 [DEBUG] Message queue: ['H', 'S']
2016/12/03 21:39:59 [DEBUG] ## process_msg ## bridge:
2016/12/03 21:39:59 [DEBUG] Message queue: ['H', 'S']
2016/12/03 21:40:00 [DEBUG] ## process_msg ## bridge:
2016/12/03 21:40:00 [DEBUG] Message queue: ['H', 'S']
2016/12/03 21:40:00 [DEBUG] ## process_msg ## bridge:
2016/12/03 21:40:00 [DEBUG] Message queue: ['H', 'S']
2016/12/03 21:40:00 [DEBUG] ## process_msg ## bridge:
2016/12/03 21:40:00 [DEBUG] Message queue: ['H', 'S']
2016/12/03 21:40:00 [DEBUG] ## process_msg ## bridge:
`

autopower, 2016-12-03:
target yun?

bilbolodz, 2016-12-03:
Generally yes, in details: Iduino Yun Shield (http://www.geeetech.com/wiki/index.php/Iduino_Yun_Shield).

autopower, 2016-12-03:
V225 onsite :)

bilbolodz, 2016-12-03:
No yet?

autopower, 2016-12-04:
Please reopen if you have problems.

closed 2016-12-04 by autopower

bilbolodz, 2016-12-04:
Not working, after putting http://XXX/data/put/profile/temp script quits with an error:
`Traceback (most recent call last):
  File "/root/thermeq3/nsm.py", line 59, in <module>
    if t3.intervals() == 0xFF:
  File "/root/thermeq3/thermeq3.py", line 617, in intervals
    self._do_beta()
  File "/root/thermeq3/thermeq3.py", line 726, in _do_beta
    self._do_profiles()
  File "/root/thermeq3/thermeq3.py", line 712, in _do_profiles
    sm, am, kk = profiles.do(self.setup.selected_mode, self.var.act_mode_idx, self.var.situation)
  File "/root/thermeq3/profiles.py", line 128, in do
    kv = temp_mode()
  File "/root/thermeq3/profiles.py", line 97, in temp_mode
    kv = table.temp[k]
TypeError: list indices must be integers, not list`

Yahoo WOEID seems to be correct because in log I can see:
`2016/12/04 22:46:42 [INFO] Difference between Yahoo and OWM temperatures. Yahoo=-1.0 OWM=3.0
2016/12/04 22:46:42 [INFO] Current temperature in Lodz is -1, humidity 90%
2016/12/04 22:46:42 [DEBUG] OWW interval updated to [420, 1260, 1260]`

which is correct name for my localisation

autopower, 2016-12-04:
V226 is onsite

bilbolodz, 2016-12-05:
Now it's working:
`2016/12/05 10:08:51 [INFO] Switching temp mode to -10 ~ 0`

autoupdate feature is working as well. Thanks.