autopower/thermeq3 #17: Saving thermeq3.bridge to internal flash

bilbolodz, 2016-12-12:
Is it a good idea to save thermeq3.bridge file on internal flash? It has "limited" number of writes (probably there is no anti wearing controller on Yun). Maybe it should be
a) moved to SD card
b) "saved in memory" at from time to time synced to other media (SD card?)

autopower, 2016-12-12:
you can simple move it to the SD card, just find out this line in thermeq3.py:
'self.bridge_file = "/root/" + self.devname + ".bridge"'
and change to path you need.

bilbolodz, 2016-12-12:
Yes I know but I'm talking about "general solution" (without needs to change code after upgrade). Maybe it's enough to make a symbolic link. How long are you running your software on Yun? Have you ever problems with errors on internal flash?

autopower, 2016-12-12:
just from the beginning of the project, its about 3 years?
Where is the /var/log stored? Logs are heavily modified files, if logs are stored on internal flash, so I don't expect problems with /root directory. Am I wrong?

bilbolodz, 2016-12-12:
Are you sure about logs:
root@Arduino:~# ls -l /var/log/
-rw-r--r--    1 root     root             0 Dec 11 20:32 lastlog
-rw-r--r--    1 root     root             0 Dec 11 20:32 wtmp
?
(fresh Yun install just thermeq3 software)

From: https://www.arduino.cc/en/Guide/ArduinoYun

`External storage on the Yún
You're discouraged from using the Yún's built-in non-volatile memory, because it has a limited number of writes. `

bilbolodz, 2016-12-12:
Maybe it's better to put it on tmpfs (/tmp) and save it "backup" at SD card?

autopower, 2016-12-12:
I'm not sure, just brainstorming

bilbolodz, 2016-12-12:
I I've made a symlink but it's also possible move whole / filesystem on SD card:
https://bitbucket.org/unitn/yuntricks/wiki/Setup%20Overlay%20filesystem

closed 2016-12-12 by bilbolodz