autopower/thermeq3 #18: Nice/status web page

bilbolodz, 2016-12-14:
It could be a good idea (in my opinion) to add to web page (nice.html status.xml) information about heating/not heating state. It allow to collect and (plot) information about "whole heating system" by external program (e.x. mrtg).

bilbolodz, 2016-12-24:
It could be also a good idea to signal (at web page) that ECO mode (e.x. using ECO button) is on
It could be also a good idea to implement signaling ECO mode ON at arduino board (using additional LED). . I've switched ECO mode accidentally a few times and I was wondering why it's so cold in the house :-)

autopower, 2017-10-07:
Please take a look at [simple dashboard](https://github.com/autopower/thermeq3/tree/master/install/dashboard)

bilbolodz, 2017-10-10:
I've tried. Generally it's working BUT dashboard claims that "Weather in Bratislava ....." but my location is other ("yahoo_location": "505120" "owm_location": "3104132"). I've looked into code and looks thats dashboard is trying to read http://localhost:8180/location.json and in case of failure silently assume Bratislava as default (

`try:
    location = d["location"]
except:
    location = "823123"
`
In my setup there are NO file location.json and that's the problem.

I also noticed that config_me.py ignores my Yahoo location ID  and puts Bratislawa ID into config file.
I've edited config file "by hand" and now nsm.py knows my correct location.

bilbolodz, 2017-10-10:
Where location.json should be created by install script? Today I've tried "upgrade" script and file still doesn't exists.

bilbolodz, 2017-10-10:
Found location.json in root folder. Unfortunately it contains :"yahoo_location" json variable but dashboard.py expects "location".

autopower, 2017-10-10:
fixed

bilbolodz, 2017-10-13:
Working fine.

closed 2017-10-13 by bilbolodz