autopower/thermeq3 #19: Crash course in configuration please?

mph070770, 2017-01-03:
Hi All,
This looks like a great project and I've purchased a Yun to try and get it working. I've been using FHEM and EQ-3 for some time but would like to just use the MAX mobile app and a boiler control - I assume thermeq3 can do this for me?

I have the sketch and linux code running. I have information in thermeq3.bridge that shows that the Yun has found by cube and the radiator valves. I haven't connected the relay yet as I want to debug my solution before I commit to installing it on my boiler. I'm not sure what I do next....

So, some questions:

1. Do I need to modify any python settings to get the system configured?
2. How do I see the status of the system. It looks like I should be using the section in the readme - "what can I change - in python code" but I'm not sure what to do? Can this be done from a browser? I tried this:

http://192.168.1.112/data/get/status

and it just responded with some JSON of:

{"value":"idle","key":"status","response":"get"}

... or should the python be accessed from a command line / telnet session?

3. Am I right to think I use the MAX SW to control the heating profile etc?

Thanks

autopower, 2017-01-04:
ad:
1) just modify config.py
2) readme updated, you don't need to modfiy anything in python code

If thermeq is correctly installed, than response from get is more verbose. I'll post sample.

mph070770, 2017-01-04:
Thanks autopower. All working now - I'd realised from reading up on the Yun that I had to access the data using the bridge and finally found the way to do it - just as you updated the readme :-)  The readme makes more sense now.

Thanks for your support. I'm about to disconnect FHEM and connect up for real. Wish me luck!

closed 2017-01-04 by mph070770