autopower/thermeq3 #20: Problem sending email

bilbolodz, 2017-10-10:
After upgrade to to 252 (also 254) there a problem with sending email. In previous versions there was a parameter self.mailuser now authentication is done using from_addr parameter. Unfortunately my mail server require real user name NOT email address to authenticate.

autopower, 2017-10-10:
Its just a renamed variable, you can use any loginname here. If not working can you please send me log?

bilbolodz, 2017-10-10:
I'm afraid that you are wrong. These variable is also used as a "source email address".
Log from nsm.py:
`2017/10/10 16:52:00 [DEBUG] Error during sending email. Error code: (501, '<bilbo_thermeq3>: sender address must contain a domain', 'bilbo_thermeq3')`

bilbo_thermeq3 = my user name for auth

autopower, 2017-10-10:
everything after Error code: comes from server, so IMO server needs domain
If auth failt, log contains: [DEBUG] Authentication error during sending email.

bilbolodz, 2017-10-10:
Yes that's correct. Your are using from_addrs variable for:
1) Login to server:
mailer.py:            server.login(eq3_setup.from_addr, eq3_setup.from_pwd)
2) Send mail message:
mailer.py:            server.sendmail(eq3_setup.from_addr, str_to_list(eq3_setup.to_addr), message)

in first case from_addr needs to be mailhost username
in second from_addr needs to be email address "send from"

It should be two different config variables as it was used to.

autopower, 2017-10-10:
Clear, will be fixed.

autopower, 2017-10-10:
Fixed.

closed 2017-10-10 by autopower