autopower/thermeq3 #21: valve_pos is not persistent over reboot

bilbolodz, 2017-10-10:
I'm setting valve_pos value using http://X.X.X.X/data/put/valve_pos/10 command. It's accepted and used in heating calculation. Unfortunately after board reset (Yun reboot) value is set again to 35% (default value).

autopower, 2017-10-10:
Is valve_pos synchronized into .bridge file located on SD card (e.g. /mnt/sda1/<name>.bridge)? If so, after reboot, this bridge file is processed and correct (taken from bridge) value is set into internal variable.
If not, please edit bridge file manually and restart python code (http://ardu.ip/data/put/msg/R) and check log file.

bilbolodz, 2017-10-10:
After Yun restart:
http://192.168.105.16/data/get/valve_pos
`value | "35"
key | "valve_pos"
response | "get"`
Request:
http://192.168.105.16/data/put/valve_pos/10
`value | "10"
key | "valve_pos"
response | "put"`
http://192.168.105.16/data/get/valve_pos
`value | "10"
key | "valve_pos"
response | "get"`
Log file:
2017/10/10 20:30:42 [INFO] idle, 10% at 2 valve(s). Checking #90 sec

cat /mnt/sda1/www/bridge.json: [..] "valve_pos": "10"
cat /mnt/sda1/thermeq3.bridge: [..] "valve_pos": "10"

Fine working.....

reboot

2017/10/10 20:36:31 [INFO] idle, 35% at 2 valve(s). Checking #90 sec
cat /mnt/sda1/www/bridge.json: [..] "valve_pos": "35"
cat /mnt/sda1/thermeq3.bridge: [..] "valve_pos": "35"

?????????????????????????????????

autopower, 2017-10-11:
I'll modify code to diagnose this.

autopower, 2017-10-11:
Fixed, correct variable name is valve_switch, bridge.py codeword dict was faulty.

bilbolodz, 2017-10-12:
NOT working:

`2017/10/12 13:08:21 [INFO] Started with PID=1603
2017/10/12 13:08:21 [INFO] Platform: yun
2017/10/12 13:08:22 [DEBUG] Bridge file loaded.
2017/10/12 13:08:22 [INFO] Hard coded ignored valves are already in bridge.
2017/10/12 13:08:25 [INFO] Mail server and credentials OK.
2017/10/12 13:08:25 [INFO] Using csv_value=3
2017/10/12 13:08:29 [DEBUG] Queueing [S]
2017/10/12 13:08:30 [DEBUG] Public IP address: 192.162.149.178
2017/10/12 13:08:36 [DEBUG] Logfile: /mnt/sda1/thermeq3_error.log
2017/10/12 13:08:36 [DEBUG] Zero sized stderr log file, nothing'll be send
2017/10/12 13:08:37 [INFO] Available file: thermeq3.zip, V252 with hash 24c892aa8e8b9f5533e56324888f1956
2017/10/12 13:08:37 [INFO] Actual version: 256, hash: 5568fb80277e351c93f0b22e4af5c1dd
2017/10/12 13:08:37 [DEBUG] Update is not necessary.
2017/10/12 13:08:39 [DEBUG] Public IP address: 192.162.149.178
2017/10/12 13:08:39 [INFO] Difference between Yahoo and OWM temperatures. Yahoo=15.0 OWM=17.0
2017/10/12 13:08:39 [INFO] Current temperature in Lodz is 15, humidity 73%
2017/10/12 13:08:39 [DEBUG] OWW interval updated to [1500, 4500, 4500]
2017/10/12 13:08:40 [DEBUG] Valve ignore interval updated to 62
2017/10/12 13:08:51 [INFO] idle, 35% at 2 valve(s). Checking #90 sec
2017/10/12 13:08:51 [INFO] Actual positions:
Room: Dominik pokoj
	[100C64] Zawor Dominik       @  0% @ 17.0'C #  0.0'C (+)
Room: Sypialnia
	[134237] Zawor Sypialnia     @  0% @ 18.0'C # 20.5'C (+)
Room: Piotr pokoj
	[134215] Zawor Piotr         @  0% @ 17.0'C #  0.0'C (+)
Room: Lazienka
	[100C8A] Zawor lazienka      @  0% @ 17.0'C #  0.0'C (+)
Room: Przedpokoj
	[13EC73] Zawor przedpokoj    @  0% @ 17.0'C #  0.0'C (+)
Room: tmp
Room: Salon, average: 0%
	[140D03] Zawor lewy          @  0% @ 18.0'C # 20.9'C (+)
	[13EDAA] Zawor srodkowy      @  0% @ 18.0'C # 20.9'C (+)
	[140CBF] Zawor prawy         @  0% @ 18.0'C # 20.9'C (+)
2017/10/12 13:08:53 [DEBUG] Bridge file (/mnt/sda1/www/bridge.json) saved.
2017/10/12 13:09:00 [DEBUG] Bridge file (/mnt/sda1/thermeq3.bridge) saved.
2017/10/12 13:09:00 [DEBUG] Message queue: ['S']
2017/10/12 13:09:00 [DEBUG] Sending message [S]

http://192.168.105.16/data/put/valve_switch/15

2017/10/12 13:10:40 [INFO] idle, 15% at 2 valve(s). Checking #90 sec
2017/10/12 13:10:40 [INFO] Actual positions:
Room: Dominik pokoj
	[100C64] Zawor Dominik       @  0% @ 17.0'C #  0.0'C (+)
Room: Sypialnia
	[134237] Zawor Sypialnia     @  0% @ 18.0'C # 20.5'C (+)
Room: Piotr pokoj
	[134215] Zawor Piotr         @  0% @ 17.0'C #  0.0'C (+)
Room: Lazienka
	[100C8A] Zawor lazienka      @  0% @ 17.0'C #  0.0'C (+)
Room: Przedpokoj
	[13EC73] Zawor przedpokoj    @  0% @ 17.0'C #  0.0'C (+)
Room: tmp
Room: Salon, average: 0%
	[140D03] Zawor lewy          @  0% @ 18.0'C # 20.9'C (+)
	[13EDAA] Zawor srodkowy      @  0% @ 18.0'C # 20.7'C (+)
	[140CBF] Zawor prawy         @  0% @ 18.0'C # 20.7'C (+)
2017/10/12 13:10:42 [DEBUG] Bridge file (/mnt/sda1/www/bridge.json) saved.
2017/10/12 13:10:48 [DEBUG] Bridge file (/mnt/sda1/thermeq3.bridge) saved.

reboot

2017/10/12 13:13:29 [INFO] Started with PID=1600
2017/10/12 13:13:29 [INFO] Platform: yun
2017/10/12 13:13:30 [DEBUG] Bridge file loaded.
2017/10/12 13:13:31 [INFO] Hard coded ignored valves are already in bridge.
2017/10/12 13:13:33 [INFO] Mail server and credentials OK.
2017/10/12 13:13:33 [INFO] Using csv_value=3
2017/10/12 13:13:38 [DEBUG] Queueing [S]
2017/10/12 13:13:38 [DEBUG] Public IP address: 192.162.149.178
2017/10/12 13:13:44 [DEBUG] Logfile: /mnt/sda1/thermeq3_error.log
2017/10/12 13:13:45 [DEBUG] Zero sized stderr log file, nothing'll be send
2017/10/12 13:13:46 [INFO] Available file: thermeq3.zip, V252 with hash 24c892aa8e8b9f5533e56324888f1956
2017/10/12 13:13:46 [INFO] Actual version: 256, hash: 5568fb80277e351c93f0b22e4af5c1dd
2017/10/12 13:13:46 [DEBUG] Update is not necessary.
2017/10/12 13:13:47 [DEBUG] Public IP address: 192.162.149.178
2017/10/12 13:13:48 [INFO] Current temperature in Lodz is 15, humidity 73%
2017/10/12 13:13:48 [DEBUG] OWW interval updated to [1500, 4500, 4500]
2017/10/12 13:13:49 [DEBUG] Valve ignore interval updated to 62
2017/10/12 13:14:00 [INFO] idle, 35% at 2 valve(s). Checking #90 sec
2017/10/12 13:14:00 [INFO] Actual positions:
Room: Dominik pokoj
	[100C64] Zawor Dominik       @  0% @ 17.0'C #  0.0'C (+)
Room: Sypialnia
	[134237] Zawor Sypialnia     @  0% @ 18.0'C # 20.5'C (+)
Room: Piotr pokoj
	[134215] Zawor Piotr         @  0% @ 17.0'C #  0.0'C (+)
Room: Lazienka
	[100C8A] Zawor lazienka      @  0% @ 17.0'C #  0.0'C (+)
Room: Przedpokoj
	[13EC73] Zawor przedpokoj    @  0% @ 17.0'C #  0.0'C (+)
Room: tmp
Room: Salon, average: 0%
	[140D03] Zawor lewy          @  0% @ 18.0'C # 20.9'C (+)
	[13EDAA] Zawor srodkowy      @  0% @ 18.0'C # 20.9'C (+)
	[140CBF] Zawor prawy         @  0% @ 18.0'C # 20.9'C (+)
2017/10/12 13:14:02 [DEBUG] Bridge file (/mnt/sda1/www/bridge.json) saved.
2017/10/12 13:14:08 [DEBUG] Bridge file (/mnt/sda1/thermeq3.bridge) saved.	`

autopower, 2017-10-12:
Please check V257

bilbolodz, 2017-10-13:
Now it's working correctly.

bilbolodz, 2017-10-14:
These time "profile" is not persistent, changed to temp after reboot again normal.

autopower, 2017-10-14:
V262, little bit tested

bilbolodz, 2017-10-14:
Setting: http://192.168.105.16/data/put/profile/temp
'profile | "temp"`

reboot

NO profile key in http://192.168.105.16/data/get/

autopower, 2017-10-14:
V263

bilbolodz, 2017-10-14:
NO profile key in http://192.168.105.16/data/get/ :-(

autopower, 2017-10-15:
check V265

bilbolodz, 2017-10-16:
Working

closed 2017-10-16 by bilbolodz