autopower/thermeq3 #23: Feature request: add more information to status.json

bilbolodz, 2017-10-11:
I'm going to put all information from EQ3 Cube to Domoticz (so far just for information/analyze). Now I'm using lua script which gets these information directly from Cube. Unfortunately Cube is "one IP connection ready" and when thermeq3 and lua script (from time to time) are running same time I've got timeout on thermeq3 or script. I'm going to write script which takes information from status.json and feeds Domoticz. Now in status.json there are not all information about devices (lack info about thermostats set temperature, no info about low battery in windows sensors or ECO button). Is it possible to put all these information in status.json? LUA script attached just for information.
[maxscript.zip](https://github.com/autopower/thermeq3/files/1375068/maxscript.zip)

autopower, 2017-10-12:
Complete dump from cube in file on port 8180 will be enough? Its an easy step, more can be implemented in future.

bilbolodz, 2017-10-12:
What format? JSON (better) or "RAW EQ3" (very hard to parse). Complete information it's very good idea because I'm also interested in RSSI and battery level. If it's easy please do it and we will see what could be done.

autopower, 2017-10-12:
far simple way: base64 decode of raw response from cube
simple way: add some info to "status" in bridge, which one? just now, there is battery info. In which response I can find RSSI?

bilbolodz, 2017-10-12:
"base64 decode of raw response from cube" is a little "to simple" :+1: -)
About RSSI. It looks that there is NO info about RSSI (sorry I thought that there is).
According: https://github.com/Bouni/max-cube-protocol/blob/master/L-Message.md
there infos about battery could be OK/LOW, about RF link OK/ERROR

autopower, 2017-10-12:
its already there, look at the bridge file in www.
https://github.com/autopower/thermeq3/wiki/system_status-in-bridge-file

bilbolodz, 2017-10-12:
I know but there no (for example) info about wall thermostat or ECO button (about eco button probably there will be only info about link and battery). There is information about window sensors but "aggregated" (mean: if there ANY windows open in a room). In one room I've two (actually I should have even more) window sensors. Maybe it's a good idea to present such complete information in bridge.json and dashboard. Even if part of information is NOT used by thermeq3 in boiler calculation it's probably a good idea to have one device (thermeq3) connecting to EQ3 Cube (and than working as such "proxy") then a few which are chasing for network access. The best idea is to use MQTT instead of WWW and json but it's beyond scope of these project and CPU in Yun is to weak for such task.

bilbolodz, 2017-10-12:
I've first alpha version of node-red script which gets information from bridge.json and put it (via MQTT) to Domoticz. Question: what's in 4 i5 filed of system_status valve table? If you consider my suggestion about changing bridge.json it could a good idea to add "decoded" separate fields for values now coded into mode [6] field.

autopower, 2017-10-12:
What you mean by "4 i5 field"?

bilbolodz, 2017-10-12:
cat /mnt/sda1/www/bridge.json
{"app_uptime": [......]
"system_status": "{ \"Salon\": {\"140CBF\": [\"Zawor prawy\", \"0\", \"21.0\", \"21.3\", \"1\", \"0\", \"24\"],
[] - Salon - room name
[] - 140CBF - RF address
[0] - Zawor prawy - valve name
[1] - 0 - percent
[2] - current temp 
[3] - set thermostat temp
[4] - 1 - ????
[5] - 0 - ????
[6] - 24 - mode

bilbolodz, 2017-10-12:
And what's the meaning of:
`"command": "", 
"heattime": "{'12-10-2017': [9058, 1507837404.816252]}", 
"ignore_opened": "53",`
?

autopower, 2017-10-12:
For other infos [take a look here](https://github.com/autopower/thermeq3/wiki/Modyfing-thermeq3)

bilbolodz, 2017-10-12:
Thank. According heatime: in bridge there is a statement:
`"heattime": "{'12-10-2017': [15240, 1507843594.7889832]}"`

but javascript (node-red) is complaining about "unexpected '". Are you sure that these statement should be:
`"heattime": "{\"12-10-2017\": [15240, 1507843594.7889832]}"`?

bilbolodz, 2017-10-13:
I've "nice" version of node-red script which takes values from bridge.json at out it (via MQTT) to Domoticz. Interested? If it's possible I'm waiting for information about wall thermostat and window sensor.

bilbolodz, 2017-10-13:
To summarize: to present "all data" from Cube in Domoticz, I need information about battery and RF link for:
-wall thermostat
-windows sensor
- ECO switch

all other information I've managed to get from current version of your bridge file. I think it could be also problem with "heating time" but I've to do more test.

bilbolodz, 2017-10-14:
According heating time: Domoticz expects continuous (over time) values for counters. When your reset heating time each day there is a problem "just after midnight" because of "internal Domoticz value overflow" (Date: 2017-10-14 00:00:00 Value: 18446744073709552000). Is it possible to add one additional value to heating table with "total (continous) heating time"?

autopower, 2017-10-14:
V259 in alpha but not tested
"dump" in bridge contains complete informations like status

bilbolodz, 2017-10-14:
Not working (quits after reading of valve). Some logs:
`2017/10/14 16:08:06 [INFO] Started with PID=3176
2017/10/14 16:08:06 [INFO] Platform: yun
2017/10/14 16:08:09 [DEBUG] Bridge file loaded.
2017/10/14 16:08:09 [INFO] Hard coded ignored valves are already in bridge.
2017/10/14 16:08:13 [INFO] Mail server and credentials OK.
2017/10/14 16:08:13 [INFO] Using csv_value=3
2017/10/14 16:08:13 [DEBUG] Queueing [S]
2017/10/14 16:08:13 [DEBUG] Public IP address: 192.162.149.178
2017/10/14 16:08:19 [DEBUG] Logfile: /mnt/sda1/thermeq3_error.log
2017/10/14 16:08:19 [DEBUG] Zero sized stderr log file, nothing'll be send
2017/10/14 16:08:20 [INFO] Available file: thermeq3.zip, V252 with hash 24c892aa8e8b9f5533e56324888f1956
2017/10/14 16:08:20 [INFO] Actual version: 259, hash: ea00cacbccec5b623fd139ccab1c6c46
2017/10/14 16:08:20 [DEBUG] Update is not necessary.
2017/10/14 16:08:21 [DEBUG] Public IP address: 192.162.149.178
2017/10/14 16:08:22 [INFO] Current temperature in Lodz is 14, humidity 79%
2017/10/14 16:08:22 [DEBUG] OWW interval updated to [1320, 3960, 3960]
2017/10/14 16:08:23 [DEBUG] Valve ignore interval updated to 60
2017/10/14 16:08:34 [INFO] idle, 25% at 2 valve(s). Checking #90 sec
2017/10/14 16:08:34 [INFO] Actual positions:
Room: Dominik pokoj
	[100C64] Zawor Dominik       @ 37% @ 21.0'C #  0.0'C (-) till 14/10/2017 17:38:31
Room: Sypialnia
	[134237] Zawor Sypialnia     @  0% @ 19.0'C # 20.5'C (+)
Room: Piotr pokoj
	[134215] Zawor Piotr         @ 17% @ 20.0'C #  0.0'C (+)
Room: Lazienka
	[100C8A] Zawor lazienka      @ 98% @ 22.0'C #  0.0'C (+)
Room: Przedpokoj
	[13EC73] Zawor przedpokoj    @  0% @ 18.0'C #  0.0'C (+)
Room: tmp
Room: Salon, average: 0%
	[140D03] Zawor lewy          @  0% @ 21.0'C # 21.1'C (+)
	[13EDAA] Zawor srodkowy      @  0% @ 21.0'C # 21.1'C (+)
	[140CBF] Zawor prawy         @  0% @ 21.0'C # 21.1'C (+)
2017/10/14 16:08:35 [DEBUG] ## control ## mode=True, heat=3, bool(heat)=True
2017/10/14 16:08:35 [INFO] heating due single valve position, no matter what 75%
2017/10/14 16:08:35 [DEBUG] Queueing [H]
`

cat thermeq3_error.log

``Traceback (most recent call last):
  File "thermeq3/nsm.py", line 7, in <module>
    thermeq3.loop()
  File "/root/thermeq3/thermeq3.py", line 943, in loop
    if t3.intervals == 0xFF:
  File "/root/thermeq3/thermeq3.py", line 432, in intervals
    self._do_thermeq()
  File "/root/thermeq3/thermeq3.py", line 501, in _do_thermeq
    self.write_strings()
  File "/root/thermeq3/thermeq3.py", line 862, in write_strings
    bridge.put("dump", self.eq3.json_dump())
  File "/root/thermeq3/maxeq3.py", line 668, in json_dump
    room_id = self.get_room_name(k)
  File "/root/thermeq3/maxeq3.py", line 131, in get_room_name
    rn = self.rooms[str(self.devices[k][3])][0]
KeyError: '0'

autopower, 2017-10-14:
V260 with try/except
if key no in devices error then room_name will be "no key" + dev_id
if  error and  key in device then room_name will be "key error" + complete device for key

bilbolodz, 2017-10-14:
root@Arduino:~# cat /mnt/sda1/thermeq3_error.log 
Traceback (most recent call last):
  File "/root/thermeq3/nsm.py", line 7, in <module>
    thermeq3.loop()
  File "/root/thermeq3/thermeq3.py", line 943, in loop
    if t3.intervals == 0xFF:
  File "/root/thermeq3/thermeq3.py", line 432, in intervals
    self._do_thermeq()
  File "/root/thermeq3/thermeq3.py", line 501, in _do_thermeq
    self.write_strings()
  File "/root/thermeq3/thermeq3.py", line 862, in write_strings
    bridge.put("dump", self.eq3.json_dump())
  File "/root/thermeq3/maxeq3.py", line 681, in json_dump
    current[room_id].update({str(k): [str(self.devices[k][2]), str(v[0]),
KeyError: "Key error: [5, 'KEQ0520581', 'Przycisk Eco wej\\xc5\\x9bcie', 0, 0, datetime.datetime(2017, 10, 14, 16, 43, 25, 787734), 16, 18, 7]"

autopower, 2017-10-14:
it's eco button, I'll take care of this

autopower, 2017-10-14:
V261
if room is 0 then "No room" is added do room list

bilbolodz, 2017-10-14:
root@Arduino:~# cat /mnt/sda1/thermeq3_error.log 
Traceback (most recent call last):
  File "/root/thermeq3/nsm.py", line 7, in <module>
    thermeq3.loop()
  File "/root/thermeq3/thermeq3.py", line 943, in loop
    if t3.intervals == 0xFF:
  File "/root/thermeq3/thermeq3.py", line 432, in intervals
    self._do_thermeq()
  File "/root/thermeq3/thermeq3.py", line 501, in _do_thermeq
    self.write_strings()
  File "/root/thermeq3/thermeq3.py", line 862, in write_strings
    bridge.put("dump", self.eq3.json_dump())
  File "/root/thermeq3/maxeq3.py", line 676, in json_dump
    current[room_id].update({str(k): [str(self.devices[k][2]), str(v[0]),
KeyError: 'No room'

autopower, 2017-10-14:
V261 again :)

bilbolodz, 2017-10-14:
Better:
`"dump": "{\"tmp\": {\"005D78\": [\"Kontaktron okienny 1\", \"4\", \"IEQ0157356\", \"Kontaktron okienny 1\", \"1\", \"0\", \"0\"]}, \"No room\": {\"0985EA\": [\"Przycisk Eco wej\\u015bcie\", \"5\", \"KEQ0520581\", \"Przycisk Eco wej\\u015bcie\", \"1\", \"0\", \"16\"]}, \"Salon\": {\"005CB5\": [\"Taras tyl\", \"4\", \"IEQ0157505\", \"Taras tyl\", \"1\", \"0\", \"16\"], \"140CBF\": [\"Zawor prawy\", \"1\", \"MEQ1459490\", \"Zawor prawy\", \"1\", \"0\", \"24\"], \"140D03\": [\"Zawor lewy\", \"1\", \"MEQ1457815\", \"Zawor lewy\", \"1\", \"0\", \"24\"], \"13EDAA\": [\"Zawor srodkowy\", \"1\", \"MEQ1450675\", \"Zawor srodkowy\", \"1\", \"0\", \"24\"], \"1678CD\": [\"Termostat scienny salon\", \"3\", \"NEQ1201071\", \"Termostat scienny salon\", \"1\", \"0\", \"24\"], \"00230E\": [\"Okno kuchnia\", \"4\", \"IEQ0161808\", \"Okno kuchnia\", \"1\", \"0\", \"16\"]}, \"Piotr pokoj\": {\"134215\": [\"Zawor Piotr\", \"2\", \"MEQ1398100\", \"Zawor Piotr\", \"1\", \"0\", \"24\"], \"14DD83\": [\"Okno Piotr\", \"4\", \"NEQ0163757\", \"Okno Piotr\", \"1\", \"0\", \"16\"]}, \"Dominik pokoj\": {\"100C64\": [\"Zawor Dominik\", \"1\", \"LEQ1128382\", \"Zawor Dominik\", \"0\", \"14/10/2017 17:38:31\", \"24\"], \"0FB2B4\": [\"Dominik okno\", \"4\", \"LEQ1112884\", \"Dominik okno\", \"1\", \"0\", \"16\"]}, \"Lazienka\": {\"100C8A\": [\"Zawor lazienka\", \"1\", \"LEQ1128403\", \"Zawor lazienka\", \"1\", \"0\", \"24\"]}, \"Przedpokoj\": {\"13EC73\": [\"Zawor przedpokoj\", \"1\", \"MEQ1452095\", \"Zawor przedpokoj\", \"1\", \"0\", \"24\"]}, \"Sypialnia\": {\"134237\": [\"Zawor Sypialnia\", \"2\", \"MEQ1398134\", \"Zawor Sypialnia\", \"1\", \"0\", \"24\"], \"16864B\": [\"Termostat scienny sypialnia\", \"3\", \"NEQ1203944\", \"Termostat scienny sypialnia\", \"1\", \"0\", \"24\"], \"0FAC8A\": [\"Okno sypialnia\", \"4\", \"LEQ1108703\", \"Okno sypialnia\", \"1\", \"0\", \"16\"]}}"`

closed 2017-10-14 by autopower