autopower/thermeq3 #24: Feature request: posibility to send to Cube "mode AUTO/ECO" command

bilbolodz, 2017-10-13:
It could be a good idea (in my opinion) to have possibility to send (using thermeq3) command set AUTO/ECO/COMFORT to cube. I've ECO button but it's available only "physically on site". I'd like to put house in ECO mode online. There are a few utilities which connects to Cube and could to that but it's always problem with "chasing for network cube access" with thermeq3.

autopower, 2017-10-13:
Setting Eco mode via eq3 portal is unusable? Or you prefer just one communication channel?

bilbolodz, 2017-10-13:
I prefer just one communication channel. Additionally 1) EQ3 portal is free only first year 2) I don't like devices which "magically" are managed via external cloud services. I'm IT guy and I know that IOT cloud devices are very buggy and it's not a good idea to let them talk "to the world".

autopower, 2017-10-14:
Please find out how to "push" eco button virtually and I'll implement such button in dashboard.

bilbolodz, 2017-10-14:
You mean: which should be send to EQ3 cube?

autopower, 2017-10-14:
yes