autopower/thermeq3 #25: Dasboard enhancement

bilbolodz, 2017-10-13:
I think that "OFF" mode of radiator (set to manual and turn wheel max left) is send as mode manual and set temperature 4.5 (checked with standalone valve and wall thermostat). Maybe it's good idea to put in into dashboard?

autopower, 2017-10-14:
You mean S message, command 40?

bilbolodz, 2017-10-14:
No. When thermostat radiator is in OFF state (you can turn off thermostat completely by setting it into manula mode na truning wheel dow "bellow 5C", on display there will be "OFF"). Its reported in bridge.json as: rradiator set to manual mode and set temperature 4.5C. Is it clear now?

autopower, 2017-10-14:
so I must send s message, command 40, with manual program and temp 4.5?

bilbolodz, 2017-10-14:
Hmmm... I think that with do not understand each other.My idea was to modify dashboard to display Mode as OFF (when radiator is OFF) not manual 4.5C......

autopower, 2017-10-14:
got it :)

autopower, 2017-10-14:
Dash updated

closed 2017-10-14 by autopower

bilbolodz, 2017-10-14:
If weather_reference is "local" temperature and humidity should be displayed from local source.

autopower, 2017-10-14:
in progress

bilbolodz, 2017-10-16:
There is a problem with correctly displayed "open window". The window in one room (Sypialnia: 134237) is open but dashboard claims that is closed (tick sign instead of cross). My bridge file is:

`{"app_uptime": "0:00:20", "autoupdate": "True", "command": "", "dump": "{\"tmp\": {\"005D78\": [\"Kontaktron okienny 1\", \"4\", \"IEQ0157356\", \"Kontaktron okienny 1\", \"1\", \"0\", \"0\"]}, \"No room\": {\"0985EA\": [\"Przycisk Eco wej\\u015bcie\", \"5\", \"KEQ0520581\", \"Przycisk Eco wej\\u015bcie\", \"1\", \"0\", \"16\"]}, \"Salon\": {\"005CB5\": [\"Taras tyl\", \"4\", \"IEQ0157505\", \"Taras tyl\", \"1\", \"0\", \"16\"], \"140CBF\": [\"Zawor prawy\", \"1\", \"MEQ1459490\", \"Zawor prawy\", \"1\", \"0\", \"24\"], \"140D03\": [\"Zawor lewy\", \"1\", \"MEQ1457815\", \"Zawor lewy\", \"1\", \"0\", \"24\"], \"13EDAA\": [\"Zawor srodkowy\", \"1\", \"MEQ1450675\", \"Zawor srodkowy\", \"1\", \"0\", \"24\"], \"1678CD\": [\"Termostat scienny salon\", \"3\", \"NEQ1201071\", \"Termostat scienny salon\", \"1\", \"0\", \"24\"], \"00230E\": [\"Okno kuchnia\", \"4\", \"IEQ0161808\", \"Okno kuchnia\", \"1\", \"0\", \"16\"]}, \"Piotr pokoj\": {\"134215\": [\"Zawor Piotr\", \"2\", \"MEQ1398100\", \"Zawor Piotr\", \"1\", \"0\", \"24\"], \"14DD83\": [\"Okno Piotr\", \"4\", \"NEQ0163757\", \"Okno Piotr\", \"1\", \"0\", \"16\"]}, \"Dominik pokoj\": {\"100C64\": [\"Zawor Dominik\", \"1\", \"LEQ1128382\", \"Zawor Dominik\", \"1\", \"0\", \"24\"], \"0FB2B4\": [\"Dominik okno\", \"4\", \"LEQ1112884\", \"Dominik okno\", \"1\", \"0\", \"18\"]}, \"Lazienka\": {\"100C8A\": [\"Zawor lazienka\", \"1\", \"LEQ1128403\", \"Zawor lazienka\", \"1\", \"0\", \"24\"]}, \"Przedpokoj\": {\"13EC73\": [\"Zawor przedpokoj\", \"1\", \"MEQ1452095\", \"Zawor przedpokoj\", \"1\", \"0\", \"24\"]}, \"Sypialnia\": {\"134237\": [\"Zawor Sypialnia\", \"2\", \"MEQ1398134\", \"Zawor Sypialnia\", \"1\", \"0\", \"24\"], \"16864B\": [\"Termostat scienny sypialnia\", \"3\", \"NEQ1203944\", \"Termostat scienny sypialnia\", \"1\", \"0\", \"24\"], \"0FAC8A\": [\"Okno sypialnia\", \"4\", \"LEQ1108703\", \"Okno sypialnia\", \"1\", \"0\", \"18\"]}}", "heattime": "{'16-10-2017': [4274, 1508163799.6959192]}", "ignore_opened": "82", "ignored": "{u'': 1924991999}", "interval": "90", "local_humidity": "66.0", "local_temp": "18.4", "mode": "auto", "msg": "", "no_oww": "0", "open_window_list": "{'0FAC8A': ['1', 'Sypialnia'], '0FB2B4': ['3', 'Dominik pokoj']}", "preference": "per", "profile": "temp", "status": "idle", "svpnmw": "75", "system_status": "{\"tmp\": {}, \"Salon\": {\"140CBF\": [\"Zawor prawy\", \"0\", \"21.0\", \"21.3\", \"1\", \"0\", \"24\"], \"140D03\": [\"Zawor lewy\", \"0\", \"21.0\", \"21.3\", \"1\", \"0\", \"24\"], \"13EDAA\": [\"Zawor srodkowy\", \"0\", \"21.0\", \"21.3\", \"1\", \"0\", \"24\"]}, \"Piotr pokoj\": {\"134215\": [\"Zawor Piotr\", \"4\", \"20.0\", \"0.0\", \"1\", \"0\", \"24\"]}, \"Dominik pokoj\": {\"100C64\": [\"Zawor Dominik\", \"0\", \"12.0\", \"0.0\", \"1\", \"0\", \"24\"]}, \"Lazienka\": {\"100C8A\": [\"Zawor lazienka\", \"74\", \"25.5\", \"0.0\", \"1\", \"0\", \"24\"]}, \"Przedpokoj\": {\"13EC73\": [\"Zawor przedpokoj\", \"0\", \"18.0\", \"0.0\", \"1\", \"0\", \"24\"]}, \"Sypialnia\": {\"134237\": [\"Zawor Sypialnia\", \"0\", \"12.0\", \"20.7\", \"1\", \"0\", \"24\"]}}", "total_switch": "150", "touch": "1508169810.37", "uptime": "0:01:52", "valve_switch": "35", "valves": "2", "weather_reference": "yahoo", "wrong_key status_key": "i"}`