autopower/thermeq3 #26: Enhancement: REAL weather from "local source"

bilbolodz, 2017-10-14:
I can easily provide REAL weather data from my local sensors. I could be done at least in there ways:
1) Inject temperature (and other data?) using http://X.X.X.X/data/put/some_variable (from script in Domoticz)
2) Provide link with Domoticz API
3) I'm exporting my data to wunderground.com and it could be get from there

What do you think about such proposal?

autopower, 2017-10-14:
Feel free to modify code, I can implement 1st option into weather.py library

bilbolodz, 2017-10-14:
First option for me is fine.

autopower, 2017-10-14:
Check V258 and  new bridge values:
`local_temp`, `local_humidity`, `weather_reference`
[quickly documented](https://github.com/autopower/thermeq3/wiki/Modyfing-thermeq3)

bilbolodz, 2017-10-14:
2017/10/14 19:21:28 [INFO] Actual version: 263, hash: d28f229c6951810f0421a540199680d6
2017/10/14 19:31:53 [INFO] Current temperature in local is 0, humidity 0%

http://192.168.105.16/data/get/ :
`weather_reference "local"
local_temp  "16.4"
local_humidity "79.0"
profile "temp"`

autopower, 2017-10-14:
Please check and test V264

bilbolodz, 2017-10-14:
1) Local weather is working fine, thank you.
2) "profile" key (set to "temp") after still returns to "normal"
3) How about "continuous heating value" (details bellow).  Are you going to implement it or should I try to fix it in the import script? If you want I can open separate issue for this.

---
According heating time: Domoticz expects continuous (over time) values for counters. When your reset heating time each day there is a problem "just after midnight" because of "internal Domoticz value overflow" (Date: 2017-10-14 00:00:00 Value: 18446744073709552000). Is it possible to add one additional value to heating table with "total (continous) heating time"?
---

autopower, 2017-10-15:
Check V266, 2 fixed, 3 implemented in total_heattime bridge value

bilbolodz, 2017-10-16:
Confirm 2 fixed.

About 3: so far NO total_heattime key in bridge.json. When it should appeal in the bridge? Just after startup or should I wait longer?

bilbolodz, 2017-10-17:
Not working:
`root@Arduino:/mnt/sda1# cat thermeq3_error.log
Traceback (most recent call last):
  File "/root/thermeq3/nsm.py", line 7, in <module>
    thermeq3.loop()
  File "/root/thermeq3/thermeq3.py", line 980, in loop
    if t3.intervals == 0xFF:
  File "/root/thermeq3/thermeq3.py", line 425, in intervals
    self.update_all()
  File "/root/thermeq3/thermeq3.py", line 616, in update_all
    self.update_counters(False)
  File "/root/thermeq3/thermeq3.py", line 680, in update_counters
    self.update_total_heat(_v)
  File "/root/thermeq3/thermeq3.py", line 625, in update_total_heat
    bridge.put("tht", value + _total_heat_time)
TypeError: can only concatenate list (not "float") to list
`

autopower, 2017-10-17:
in V266?

bilbolodz, 2017-10-17:
Yes

autopower, 2017-10-17:
Fixed

bilbolodz, 2017-10-20:
Working

closed 2017-10-20 by bilbolodz