autopower/thermeq3 #27: Valve without external wall thermostat always shows 0 temperature

bilbolodz, 2017-10-16:
Valve without external wall thermostat always shows 0 temperature. Is it good situation or is it en error. I know that presentation of these values by valve to cube is rather rare but previous script sometimes reported values. When I switched to thermeq3 never got usable temperature. Of course it could be a false positive conclusion but "just asking".

autopower, 2017-10-16:
Yes, its a known problem of eq3 system.  Take a look [here](https://github.com/autopower/thermeq3/issues/5#issuecomment-197735924)
to get (sometimes obsolete) value modify code, maxeq3.py, set to False:
```
        # set zero temp from HT or not
        self.set_zero = True
```

bilbolodz, 2017-10-17:
Not working. After setting False nsm.py quits with an error:

`
Traceback (most recent call last):
  File "/root/thermeq3/nsm.py", line 6, in <module>
    thermeq3.start()
  File "/root/thermeq3/thermeq3.py", line 924, in start
    t3.prepare()
  File "/root/thermeq3/thermeq3.py", line 106, in prepare
    eq3_result, eq3_error = self.eq3.read_data(True)
  File "/root/thermeq3/maxeq3.py", line 305, in read_data
    self.read(refresh)
  File "/root/thermeq3/maxeq3.py", line 329, in read
    self.cmd_l(sd)
  File "/root/thermeq3/maxeq3.py", line 472, in cmd_l
    valve_curr_temp = self.valves[valve_adr][2]
KeyError: '140D03'
`

autopower, 2017-10-17:
fixed in V267
if on 1st read is 0.0 then 0.1 is written (to indicate that this is synthetic value)

bilbolodz, 2017-10-17:
Working

closed 2017-10-17 by bilbolodz