autopower/thermeq3 #29: Enhancement: error reporting in bridge.json

bilbolodz, 2017-10-18:
It could be a good idea to report problems with connection to EQ3 Cube in bridge.json. It will allow (using external program and hardware) to reset Cube to get it working. Now I can see in log file:
`2017/10/18 11:27:20 [DEBUG] Queueing [E]
2017/10/18 11:27:20 [ERROR] Timeout. Operation in progress.
 Timeout. Operation in progress.`

but there is no info about it in bridge.json so Domoticz is unaware of these situation.

autopower, 2017-10-22:
Check V270
bridge key status_key shows 'E' and status shows "Error: [<error description>]"

bilbolodz, 2017-10-23:
NOT working. I can see 
`status_key | "e"
status | "Error: [Timeout!]"
`
in http://192.168.105.16/data/get/ but in bridge.json stiil:
`"status": "idle", "status_key": "i"`

Some logs:
`2017/10/23 12:33:28 [DEBUG] Bridge file (/mnt/sda1/www/bridge.json) saved.
2017/10/23 12:33:35 [DEBUG] Bridge file (/mnt/sda1/thermeq3.bridge) saved.
2017/10/23 12:33:35 [DEBUG] Message queue: ['S']
2017/10/23 12:33:35 [DEBUG] Sending message [S]
2017/10/23 12:34:32 [DEBUG] Queueing [E]
2017/10/23 12:34:33 [ERROR] Timeout. Operation in progress.
 Timeout. Operation in progress.
 Timeout. Operation in progress.
 
2017/10/23 12:34:33 [DEBUG] Message queue: ['E']
2017/10/23 12:34:33 [DEBUG] Sending message [E]
2017/10/23 12:35:20 [DEBUG] Queueing [E]
2017/10/23 12:35:21 [ERROR] Timeout. Operation in progress.
 Timeout. Operation in progress.
 Timeout. Operation in progress.
[..]
2017/10/23 12:38:54 [DEBUG] Message queue: ['E']
2017/10/23 12:38:54 [DEBUG] Sending message [E]
2017/10/23 12:39:08 [DEBUG] Public IP address: 192.162.149.178
2017/10/23 12:39:09 [INFO] Current temperature in local is 9.9, humidity 87.0%
2017/10/23 12:39:09 [DEBUG] OWW interval updated to [780, 2340, 2340]
2017/10/23 12:39:09 [DEBUG] Valve ignore interval updated to 49
2017/10/23 12:39:59 [DEBUG] Queueing [E]
2017/10/23 12:40:00 [ERROR] Timeout. Operation in progress.
 Timeout. Operation in progress.
 Timeout. Operation in progress.
 Timeout. Operation in progress.
 Timeout. Operation in progress.
 Timeout. Operation in progress.
 Timeout. Operation in progress.
 Timeout. Operation in progress.
 Timeout. Operation in progress.
 Timeout. Operation in progress.
 Timeout. Operation in progress.
 Timeout. Operation in progress.
 
2017/10/23 12:40:00 [DEBUG] Message queue: ['E']
2017/10/23 12:40:00 [DEBUG] Sending message [E]
2017/10/23 12:40:39 [DEBUG] Queueing [E]
2017/10/23 12:40:40 [ERROR] Timeout. Operation in progress.`