autopower/thermeq3 #30: Problem with permanent setting profile to term

bilbolodz, 2017-10-23:
When I'm trying to set profile to "temp" there is something strange with it. I can set profile to temp using "http://192.168.105.16/data/put/profile/temp" and it's correctly set (checking http://192.168.105.16/data/get/). It even set after reboot but when then when I'm checking it after some time (hard to determine after which time) profile is set again to normal.

pokoje, 2017-11-14:
I am able to set profile to "temp" or "time" and it persists in bridge. But I do not see any changes in thermeq3's behaviour- values that should change seen are constant in dashboard , despite the change of time zone or temperature. Ver 2.52

autopower, 2018-11-24:
check this wiki page: https://github.com/autopower/thermeq3/wiki/Profiles

closed 2018-11-24 by autopower